stooged/ESP32-Server-900u #1: 'USBMSC' does not name a type

maj0r1337, 2022-01-25:
hi stooged,

ive got this error at compiling.

ESP32_Server_900u:25:1: error: 'USBMSC' does not name a type
 USBMSC dev;
 ^~~~~~
C:\Users\xxx\Downloads\ESP32-Server-900u-main\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino: In function 'void setup()':
ESP32_Server_900u:576:3: error: 'USB' was not declared in this scope
   USB.begin();
   ^~~
C:\Users\xxx\Downloads\ESP32-Server-900u-main\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino: In function 'void enableUSB()':
ESP32_Server_900u:583:3: error: 'dev' was not declared in this scope
   dev.vendorID("USB");
   ^~~
C:\Users\xxx\Downloads\ESP32-Server-900u-main\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino:583:3: note: suggested alternative: 'div'
   dev.vendorID("USB");
   ^~~
   div
C:\Users\xxx\Downloads\ESP32-Server-900u-main\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino: In function 'void disableUSB()':
ESP32_Server_900u:600:3: error: 'dev' was not declared in this scope
   dev.end();
   ^~~
C:\Users\xxx\Downloads\ESP32-Server-900u-main\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino:600:3: note: suggested alternative: 'div'
   dev.end();
   ^~~
   div
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi
 Not used: C:\Users\xxx\Downloads\arduino-1.8.19-windows\arduino-1.8.19\libraries\WiFi
exit status 1
'USBMSC' does not name a type

stooged, 2022-01-25:
You need an esp32 S2 board for this project, the original esp32 does not support the usb functions.

maj0r1337, 2022-01-25:
ah, i see. thank you

closed 2022-01-25 by maj0r1337