stooged/ESP32-Server-900u #3: QUESTION.

Doyle4, 2022-01-27:
Sorry to bother again.

I was wondering, as the exfathax now has a pico image 6kb rather than 4.1mb, could it now be possible to to add USB Emulation to this ESP? like what the Raspberry Pi Zero W has had made.
My thinking was instead of the USB stick connected, the cable powering the ESP could also send the data to the PS4?

This would be a game changer as the Zeros are out of stock everywhere and ESPs are not.

stooged, 2022-01-27:
transferred issue this to the correct repo and cleaned it to just reference the compile problems

stooged, 2022-01-27:
transferred this issue to the correct repo and cleaned it to just reference the compile problems.

Doyle4, 2022-01-27:
Thank you, will let others know.

EchoStretch, 2022-01-28:
Not sure what im doing wrong or maybe its my board but can you check this out
https://youtu.be/JMqYdiWf8RI

closed 2022-01-31 by stooged