stooged/ESP32-Server-900u #4: Wiring diagram?

subworx, 2022-01-27:
Hi,
I plan on getting two Wemos/Lolin S2 mini soon to implement this, once you confirmed it working fine.

Do you have a wiring diagram available for this setup, similar to the ones you posted for the PS4-Server-900u (which I'm going to build this weekend)? 

Thank you very much :)
Regards

stooged, 2022-01-27:
There is no wiring involved with this project.

closed 2022-01-27 by stooged

stooged, 2022-01-31:
just an update, some boards need wiring but i looked into the s2 mini and that board does not need any wiring as the usb port is connected to the native usb.

you will need to put the board into dfu mode to upload the sketch which is detailed here: 
https://www.wemos.cc/en/latest/tutorials/s2/get_started_with_arduino_s2.html