stooged/ESP32-Server-900u #5: Failed To Trigger

EchoStretch, 2022-01-28:
If yo have time can you check out my video and see what im doing wrong.. maybe its my board.. idk
https://youtu.be/JMqYdiWf8RI

stooged, 2022-01-28:
The by the look of it the usb part of the code is not working which is the part i cant test on the esp board.
The library on the samd boards is different and that is the only test board i have at the moment.
Once my esp boards arrive i should be able to test and fix the issue.

stooged, 2022-01-28:
i just pushed a change to the repo, try that and see if the usb mounts during the loading exfathax stage.

EchoStretch, 2022-01-28:
Same failed to trigger

cspencer49519, 2022-01-28:
Could this be related to the USB port being hooked up to the USB/UART bridge instead of USB OTG by default? I've got a board OTW but I was referencing this information, specifically the 'electrical connection' section:

https://blog.espressif.com/dfu-using-the-native-usb-on-esp32-s2-for-flashing-the-firmware-b2c4af3335f1

stooged, 2022-01-28:
It was worth a try.

Once my boards arrive i should be able to work out what is happening.

With the otg stuff the esp should be able to switch modes in the code.

I am not sure if all the esp32-s2 boards will work so i have ordered a few different types to test with.
The board i want to work the most is the s2 mini because of its size.

cspencer49519, 2022-01-28:
> It was worth a try.
>
> Once my boards arrive i should be able to work out what is happening.
>
> With the otg stuff the esp should be able to switch modes in the code.
>
> I am not sure if all the esp32-s2 [...]

The 'electrical connection' section of the link I posted states the following:

As you can see, the Saola board doesn’t have the connector for the native USB, only for the external USB-to-UART. To use the native interface, you need to connect a breakout USB cable as follows:

    USB D- to GPIO19
    USB D+ to GPIO20

Also, the first review under this link:

https://www.amazon.com/Unexpected-Maker-TinyS2-ESP32-S2-Board/dp/B093CJ4V3Y

States:

"If you are getting this for the "S2" part of the ESP32, the fact that it has native USB, then understand that the native USB is just some pins on the board. The USB Micro connection on the board goes through a USB/UART converter and into the normal RX/TX pins. So if you want to play with the USB part get a USB breakout board as well, or - as I did since I forgot it and could wait - waste a old USB cable and solder it to a header or directly to the board."

This review also includes an image if the connections this person used. I'm not 100% sure, but it looks like some of these boards only handle USB/UART over the included USB port, and D+, D- pins need to be connected in that scenario.

stooged, 2022-01-28:
That is what i mean about different boards may not work, the link you posted is for a tiny s2 but if you look at this board it has dip switches to select the usb mode and it works as a mass storage device.
https://youtu.be/l3MmrVHMd94

some other boards have 2x usb ports and the samd51 boards switch the mode pragmatically.
That is why i ordered a few different types.

I have these board coming for testing

Esp32 s2 saola wrover
Esp32 s2 woor(this has dip switches, same board as video)
Ttgo t8 esp32 s2 v1.1 with tf card slot
ESP32-S2-DevKitM-1
Wemos s2 mini

stooged, 2022-01-28:
I had a look at the devkitm board and it might be the same as the tiny s2 and some boards may need a usb a cable connected to the board for it to work which i will also test.

Pin 19 and 20

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/_images/esp32-s2-devkitm-1-v1-pin-layout.png

oxstretchxo, 2022-01-28:
Ok I ordered 
ESP32-S2-DevKitC-1
I might order a few more after work for testing

stooged, 2022-01-29:
yes i think the <a href=https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-s2-devkitc-1.html>ESP32-S2-DevKitC-1</a> might be the best board for most people to get to have a wiring free setup.

looking at the board schematic the second usb port interfaces with the pins 19 and 20

[image]

stooged, 2022-01-29:
based on the ESP32-S2-DevKitM-1 this would be a wiring layout for the usb

[image]

oxstretchxo, 2022-01-29:
I ordered this
https://www.digikey.ca/en/products/detail/adafruit-industries-llc/4449/11569127

But I thought you plugged in your USB device to the boatd

Doyle4, 2022-01-29:
Great to see how this is coming along.
Im interested to see which boards do work, even with some modification.
With productions slow in the current pandemic, the Pi Zero W/2 are impossible to get hold of or are priced high by scalpers, ESP boards, still many out there.
Thanks your time and work on this guys.

Doyle4, 2022-01-29:
Looks like mr crump has done it.
https://www.youtube.com/watch?v=ZfU_1_iIBIA

stooged, 2022-01-29:
> I ordered this https://www.digikey.ca/en/products/detail/adafruit-industries-llc/4449/11569127
> 
> But I thought you plugged in your USB device to the boatd

the board emulates the usb drive so no drive is needed, that part (usb to jumper) would be great if it had a usb A male port on the end.
you could still use it with a usb A male to usb A male cable

oxstretchxo, 2022-01-29:
Ya I got a few of them laying around.. trying to find the male end ones with the jumped ends so I won't have to solder

cspencer49519, 2022-01-29:
I grabbed this board:

https://www.amazon.com/dp/B08PCPP729?psc=1&smid=A28JUS3SJ1A0RV&ref_=chk_typ_imgToDp

It's an 'ESP32-S2 Saola 1R Dev Kit Featuring ESP32-S2 WROVER'. It will be here tomorrow, the pinout shows d+ and d- on 19 and 20, so I think it will work. If you want to do UART/Flashing/Native all on one interface it looks like that can work too, but you need too generate a custom configuration.

stooged, 2022-01-29:
in theory they should all work if we can pipe the USB- and USB+ to the ps4.

i cant find much info on the s2 mini board so i think the best board for most people to use that are not really good with wiring or do not have the tools to solder etc will be the <a href=https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-s2-devkitc-1.html>ESP32-S2-DevKitC-1</a> because it has both usb ports onboard so there is no switches or wiring involved it will just be a matter of changing the port they use to program the sketch and plug into the ps4

oxstretchxo, 2022-01-29:
Tired USB Plug and no luck

oxstretchxo, 2022-01-29:
<iframe title="vimeo-player" src="https://player.vimeo.com/video/671359297?h=b630a09f53" width="640" height="360" frameborder="0" allowfullscreen></iframe>

oxstretchxo, 2022-01-29:
https://player.vimeo.com/video/671359297?h=b630a09f53

stooged, 2022-01-29:
that is confusing, the only place the board should be plugged into is the usb port on the ps4.

EchoStretch, 2022-01-29:
Yea Ive Tested Both ps4 usb port and computer but nothing..

neo305, 2022-01-29:
> https://player.vimeo.com/video/671359297?h=b630a09f53

Forgive me but are you even flashing the exfathax.h file?

stooged, 2022-01-29:
Use this example sketch to test if it is working as a mass storage device 

https://github.com/espressif/arduino-esp32/tree/master/libraries/USB/examples/USBMSC

Once loaded it will show up on your pc as a drive when you plug it in if the wiring is all good.

EchoStretch, 2022-01-29:
[image]
should any of the options be changed

stooged, 2022-01-29:
> [image] should any of the options be changed

I am not sure, firmware msc on boot maybe.

neo305, 2022-01-29:
I ordered this one. Does this one require me to solder? Cuz I want the one that I don't have to solder. ESP32-S2 Saola 1R Dev Kit Featuring ESP32-S2 WROVER https://www.amazon.com/dp/B08PCPP729/ref=cm_sw_r_cp_api_glt_i_F0VY1W3N31PM4YXX1D30?_encoding=UTF8&psc=1

stooged, 2022-01-29:
> I ordered this one. Does this one require me to solder? Cuz I want the one that I don't have to solder. ESP32-S2 Saola 1R Dev Kit Featuring ESP32-S2 WROVER https://www.amazon.com/dp/B08PCPP729/ref=cm_sw_r_cp_api_glt_i_F0VY1W3N31PM4YXX1D30?_encoding=UTF8&psc=1

That does look like one that needs a plug attached.

EchoStretch, 2022-01-29:
yea flashed the usb but computer don't see anything

EchoStretch, 2022-01-29:
Yes it does come up as usb storage
[image]

neo305, 2022-01-29:
O brother :( can you tell me which one to get that doesn't require me to solder and will load the exfathax so I don't have to use a usb flash drive.

EchoStretch, 2022-01-29:
@neo305 it has the pins on the back so if it will work u could just use something like this
https://www.adafruit.com/product/4448

neo305, 2022-01-29:
> @neo305 it has the pins on the back so if it will work u could just use something like this
> 
> https://www.adafruit.com/product/4448

Thanks but no thanks lol yeah $2 with $14 shipping lol I want a board that doesn't require anything just plug and play

EchoStretch, 2022-01-29:
yea not that place.. hahah but a local computer store maybe have one for 2$

stooged, 2022-01-29:
> Yes it does come up as usb storage [image]

Ok then it is something in the code that i will need to look at when my board arrives.

stooged, 2022-01-29:
> O brother :( can you tell me which one to get that doesn't require me to solder and will load the exfathax so I don't have to use a usb flash drive.

It is best to wait until i can test the boards i have on order. 
I cant test anything until i get atleast 1 esp-32 s2 board.

GameplexSandy, 2022-01-29:
> That is what i mean about different boards may not work, the link you posted is for a tiny s2 but if you look at this board it has dip switches to select the usb mode and it works as a mass storage device through the single usb port. https://youtu.be/l3MmrVHMd94
> 
> some other boards have 2x usb ports and the samd51 boards switch the mode programmatically. That is why i ordered a few different types.
> 
> I have these boards coming for testing
> 
> Esp32 s2 saola wrover Esp32 s2 woor (this has dip switches, same board as video) Ttgo t8 esp32 s2 v1.1 with tf card slot ESP32-S2-DevKitM-1 Wemos s2 mini
> 
> i have not ordered this one but it is the one with twin usb ports [ESP32-S2-DevKitC-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-s2-devkitc-1.html)
> 
> there are many different types so it will take a bit of testing to work out which boards work. my boards with any luck start arriving next week and keep coming in over the next 2 weeks.

Hi stooged, i found this might help someone also by moving the 0 ohm resistor on the board itself.
I know it is very complex for people who are not good at soldering or using smd hot air station for moving the resistor.
 Here is the link below,
https://www.esp32.com/viewtopic.php?f=12&p=90982

And one more thing i ordered this model of esp32-s2 saola dual usb. I am sure this model also works u can add this in your working model boards support in github.
I ordered from amazon.in coz i am from india i found it here-.  Link1-https://www.amazon.in/dp/B09633B843/ref=cm_sw_r_apan_glt_i_T9WYJBPVBNAQXASWGA2Z

Same seller store with his own website link 2-. https://www.teztech.in/ESP32-S2-SAULA-MODUAL-p331270033

Link3 -https://m.aliexpress.com/item/1005003312519180.html

[image]
[image]
[image]
[image]
[image]

GameplexSandy, 2022-01-29:
> I grabbed this board:
> 
> https://www.amazon.com/dp/B08PCPP729?psc=1&smid=A28JUS3SJ1A0RV&ref_=chk_typ_imgToDp
> 
> It's an 'ESP32-S2 Saola 1R Dev Kit Featuring ESP32-S2 WROVER'. It will be here tomorrow, the pinout shows d+ and d- on 19 and 20, so I think it will work. If you want to do UART/Flashing/Native all on one interface it looks like that can work too, but you need to generate a custom configuration.
> 
> [image]

[image]
I grabbed this dual usb version

cspencer49519, 2022-01-30:
My board arrived today, I wired up the USB and attempted the exploit. I don't even get as far as echostretch got, when I click the GoldHen button it just sits there, attempting to visit the page on my computer, it looks like it stalls at the loader. I have uploaded the spiffs FS using the 'Upload spiffs data to esp32' plugin, which has worked for me with several projects in the past. I can see the payloads, they just don't seem to be loading.

oxstretchxo, 2022-01-30:
Don't use esp32fs the one in the list me-no-dev is not for s2 and it don't work. I did find another that distoryed the html and did exactly what u said. Just flash and upload payloads with website.

cspencer49519, 2022-01-30:
I spoke too soon, I flashed the device with a linux box, and I installed fresh copies of espassyncwebserver and async tcp, I now get the exploit, exfathax etc. notifications, but I'm experiencing the same issue, failed to trigger exploit.

oxstretchxo, 2022-01-30:
Yea its the code @stooged will fix it when he gets a board

cspencer49519, 2022-01-30:
I've done a little further testing, and the problem appears to be in the 'enableUSB' function. If I manually create a post request to 10.1.1.1/usbon the drive is enabled. Running the code in browser it reaches the section where 'exfathax' is displayed onscreen, where the 'enableUSB' functon is ran, the USB drive is not enabled. Also, sending a POST request to 10.1.1.1/usboff doesn't appear to disable the drive. Not 100% sure this is the issue, just an observation.

cspencer49519, 2022-01-30:
Success. I swapped out the url in the 'enableUSB' command inside loader.html, and instead set the destination to 'http://10.1.1.1/usbon'. Exfathax is now triggered, fittest run froze at loading GoldHEN with a couple graphical glitches, trying again now.

cspencer49519, 2022-01-30:
Second try worked fine! Goldhen loaded.

https://vimeo.com/671753822

cspencer49519, 2022-01-30:
Modified loader.h.

`// We need this header file to use the flash memory as storage with the PROGMEM directive:
#include <pgmspace.h>
// Byte length: 12214
constexpr uint8_t loader[] PROGMEM = {
    31, 139,   8,   8, 113,  10, 247,  97,   0,   3, 108, 111,  97, 100, 101, 114,  46, 104, 116, 109, 108,   0, 229, 125, 109, 123, 219,  54, 178, 232, 231, 230,  87, 176, 190, 119, 107, 169, 145,  37, 146, 122, 163, 226,  56, 231,  74, 178, 180, 205, 105, 211, 228, 198, 105, 247, 236, 205, 241, 213,  82,  18, 101, 177, 150,  68,  45,  73, 249, 165, 221, 252, 247,  59,  51,   0,  72,   0,   4,  37,  57,  77, 119, 207, 115, 174, 243, 196, 150, 192, 193, 188,  97,  48,  24,   0,   3, 240, 229, 215, 151, 111, 135,  31, 254, 250, 110, 100,  45, 211, 245, 234, 213, 179, 151, 236,  15, 252,  13, 252, 249, 171, 103,  95, 189,  76, 195, 116,  21, 188, 250,  33, 242, 231,  65, 252, 178, 193, 190,  65, 241,  58,  72, 125, 107, 227, 175, 131, 139, 187,  48, 184, 223,  70, 113, 106, 205, 162,  77,  26, 108, 210, 139, 147, 251, 112, 158,  46,  47, 230, 193,  93,  56,  11, 206, 232,  75, 205,  10,  55,  97,  26, 250, 171, 179, 100, 230, 175, 130,  11, 231,   4, 113,  36, 233,  35,  33, 251, 106,  26, 205,  31, 173, 223, 224, 195,  87, 179, 104,  21, 197,  47, 172, 251, 101, 152,   6, 231,  88, 176,   0, 164, 103,  73, 248, 107, 240, 194, 114, 237, 237,   3, 149, 165, 193,  67, 122, 230, 175, 194, 155, 205,  11, 107,   6,  20, 131, 152, 138, 215, 126, 124,  19,  66, 145,  77, 223, 162, 187,  32,  94, 172, 162, 251,  23, 214,  50, 156, 207, 131,  13,  22, 126, 122,   6, 191, 234, 225, 102,  17,  49, 106,  38, 152, 175, 182,  81,   2, 172,  70, 128, 104,  17,  62,   4, 115, 173, 204, 159,  38, 209, 106, 199, 153,  75, 163, 237,  11, 171, 213, 254,  19, 125,  89,   5, 139, 244, 133, 213, 182, 255,  84, 224, 187, 205, 249, 166, 178, 133, 191,  14,  87, 143,  47, 172, 196, 223,  36, 103,  73,  16, 135, 139, 115,  73, 240, 255,  49, 245, 240,  31, 195,  30,   3, 200,  34, 138, 215,  47,  44, 250, 184, 242, 211, 160, 114,   6,   4, 106,  22, 254, 174,  50, 137, 190, 122, 217,  16, 122, 124, 153, 204, 226, 112, 155, 162,  70,  27,  13,  43, 138,  67,  80, 135, 191, 178, 130, 135, 237,  42,  10, 129, 183, 101, 154, 110, 147,  23, 141, 198,  77, 152,  46, 119, 211, 250,  44,  90,  55, 134, 203,  96,  51, 143, 134,  75, 127, 219, 216, 190, 125,  59,  72,  90, 207, 158, 221, 249, 177, 181, 245,  31,  87, 208, 226, 227, 112,  21,  88,  23,  86,  18,  36,   9,   8, 127, 149,  70, 177, 127,  19, 212, 111, 130, 244, 117,  26, 172,  43, 167,  28, 234,  20,  56, 145,  42, 125,  64,  19, 217,  83, 139,  76,  72, 212, 217,  37, 211, 191, 248,  97, 250,  33,  92, 239, 171, 114,  15,  32,  41, 128, 104, 148,  46, 125, 176, 193,  11, 235, 228, 228, 252, 217, 179, 112,  97,  85, 190, 150, 176,  85, 159, 253, 246, 204, 210, 208,  59,  54, 252, 156,  91, 141, 198,  60,  88, 248, 187,  85, 106,  65, 165,  96, 189,  77,  31, 159, 129,  97,  44, 118, 155,  25,  54, 177, 149,  44, 163, 251,  55, 192,   8, 112,  80,  89,  39,  55,  85,  11,  17, 205, 163, 217, 110,  13, 166, 134,  60, 141,  86,   1, 126,  28,  60, 190, 158,  87,  78, 214,  12, 242, 164,  10, 102, 181,   9, 226, 239,  62, 188, 249,   1,  40,  65, 189, 243, 227, 106,  81, 219, 213, 231,  97, 178,  93, 249, 143,  23, 167, 211,  85,  52, 187,  61,  61,  87,  24,  66,  81, 223, 229,  18,  87, 170,  32, 129, 181, 141,   3,  44, 
    16, 170, 176, 230, 240, 132,  68,  70,  69,  72, 205,  87, 133,  34,  44, 102,  63, 168, 188, 135, 101,  12,  28, 110, 130, 123, 235,  63, 222, 252, 240,  29, 152, 196, 251, 224, 239, 187,  32,  73,  43, 213, 243,  12,  14,  96, 234, 209,  54, 216,  84,  78, 255,  60, 250, 112,  90, 147, 237, 161,   6, 198, 184,  11, 116,  88, 232,  72, 113,  56,  15, 222, 128, 162,  63,  60, 110,   3, 104, 101, 232, 163,  13, 144,  41, 220, 156,  91, 179, 165,  31,  39,  65, 122, 241, 112, 182,   3, 131,  63,   3, 237, 135, 155, 128, 204,  70, 193, 177,  33,  65,  46,  44,  33, 120,  37, 168,  74, 172, 163,  92, 233,  50,  76,  64,  99, 126, 186,  75, 172, 139,  11, 112,   8, 182,  12, 129,  63, 170, 101,  16, 124,  28,  36, 219, 104, 147,   4,  57, 181,  79, 217, 167,  96, 149,   4, 217,  23,  21, 147, 108,   6,  39,  99,  31,  36, 159,  91, 105,  68, 141,  97, 157,  88, 207, 149,  46, 242,  28,  74, 206, 168,  84, 226,  80,  18, 239, 211,  39,  85, 212,   4,  58,  29,  83,  55, 178, 194, 153, 224, 228,  21, 186,  63,  70,  89,   3, 131,   2, 235,  39, 188,  14,  88, 199,  51, 176,   1, 176, 238,  76,  89,   9,  48, 144, 164,  81, 116,  19, 204, 115, 203,   1, 179, 242, 167, 171, 224, 167, 171,  65,  69,  40, 138,  58,  93, 188, 194, 110, 131, 238,   0, 188, 129,  99, 215,  29, 252, 215,   0, 116, 209,  98, 113, 114, 158, 193, 129, 213, 110,  87, 123, 109, 133,  32, 152, 165, 156, 188, 123, 123, 245, 225, 164, 134, 200,  21,  19,  97,  32,  36, 241, 102, 183,  90,  85, 201, 182, 115,  22, 131, 205,  19,  57, 220, 252, 147,  25,  12,  55, 191,   4, 179, 148, 247,  63, 234, 123, 243,  71,  24, 247, 194,  89, 214,  50,  12,  66, 210,  63, 147,  67, 234, 136, 104, 142, 245,  85, 176, 185,  73, 151, 214,  43, 203, 174,  62,  83, 204,  77, 242, 104, 147, 233, 110, 177,   8, 176, 127,  66, 159,   9,  55, 245, 228,  17, 199, 203,  85, 165, 213, 237, 214,  72, 202, 112, 147, 118,  90,  21, 251, 193, 237, 184, 232, 204, 236, 154, 101,  63, 244, 170, 248, 187, 201, 191, 119, 241,  75, 203, 161, 207, 103,  14, 124, 201, 237,  16,   9,   1, 129,  31, 130,  13,  16,  40, 178, 118, 110,  98 [... cut ...] 11, 106,  71,  31,  95,  54, 112, 146, 241, 234, 217, 179, 151, 211, 104, 254, 104,  69,  27,  60, 254, 121, 113,  34,  29,  12, 221,  98,  10, 128, 211, 198, 227, 160,  39, 120,  84, 114,  30, 222,  89, 179, 149, 159,  36, 
    23,  39, 120, 183, 205, 137,  21,   2, 252, 154,  29,  64,  61, 121, 149,  95, 198,  68,  49, 107, 205, 122, 199, 174,  13, 194, 219, 150, 172, 122, 189, 254, 178,   1, 245, 145,  44, 146,  67, 178, 180, 201, 242, 234, 255,   1, 102, 116, 158, 245, 173, 192,   0,   0
};`

EchoStretch, 2022-01-30:
Great Job.. Gonna Test Now

cspencer49519, 2022-01-30:
> Great Job.. Gonna Test Now

Make sure you manually unhook it after you're done, I'm not sure the usboff command is working.

EchoStretch, 2022-01-30:
C:\Users\Echo Stretch\Desktop\ESP32S2\ESP32-Server-900u-main\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino: In lambda function:
ESP32_Server_900u:512:87: error: 'loader_gz' was not declared in this scope
         AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", loader_gz, sizeof(loader_gz));
                                                                                       ^~~~~~~~~
C:\Users\Echo Stretch\Desktop\ESP32S2\ESP32-Server-900u-main\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino:512:87: note: suggested alternative: 'loader'
         AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", loader_gz, sizeof(loader_gz));
                                                                                       ^~~~~~~~~
                                                                                       loader
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Echo Stretch\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
'loader_gz' was not declared in this scope

cspencer49519, 2022-01-30:
> C:\Users\Echo Stretch\Desktop\ESP32S2\ESP32-Server-900u-main\ESP32-Server-900u-main\ESP32_Server_900u\ESP32_Server_900u.ino: In lambda function: ESP32_Server_900u:512:87: error: 'loader_gz' was not [...]

I borked that paste, my apologies.

https://pastebin.com/XHgxLpmU

EchoStretch, 2022-01-30:
Strange i changed the info in loader but same error.

cspencer49519, 2022-01-30:
> Strange i changed the info in loader but same error.

That doesn't make any sense. loader_gz would be declared, can't produce the same error.

EchoStretch, 2022-01-30:
Yea i confirmed everything was correct but same error when change loader code to the one you modified.

cspencer49519, 2022-01-30:
> Yea i confirmed everything was correct but same error when change loader code to the one you modified.

Did you get the code from the second link I posted? Look at the code, loader_gz is declared. If you're getting the same error then idk, is the file saving?

EchoStretch, 2022-01-30:
[image]

cspencer49519, 2022-01-31:
> [image]

So it wrote successfully, right?

EchoStretch, 2022-01-31:
yep but same error .. this blows my mind u sure you uploaded the correct one.. hahah

cspencer49519, 2022-01-31:
> yep but same error .. this blows my mind u sure you uploaded the correct one.. hahah

Same error as in what? It gets to exfathax then times out? Did you hook up your own usb cable? Did you try the test sketch to verify that your device show up under windows as a usb mass storage controller?

EchoStretch, 2022-01-31:
Failed To Trigger Error
Yes Im running usb from pin 20 21 for power and 19 and 20 for USb
Sketch uses 755606 bytes (57%) of program storage space. Maximum is 1310720 bytes.
Global variables use 43764 bytes (13%) of dynamic memory, leaving 283916 bytes for local variables. Maximum is 327680 bytes

Ive also tested usbmsc and it comes up as mass storage device.

[22 more replies]

closed 2022-01-31 by stooged