stooged/ESP32-Server-900u #6: How to add host

neo305, 2022-01-28:
Sorry I know but where can I get help adding my host? Is there a forum or something? Do I just copy same data folder that I flash my esp8266 with?

stooged, 2022-01-28:
yes you can just upload the html and bins like the esp8266 but the project is not working properly yet so it is best to wait until i update it.

neo305, 2022-01-28:
Ok thanks. I ordered this. Will it work? Did I get the right one? ESP32-S2 Saola 1R Dev Kit Featuring ESP32-S2 WROVER https://www.amazon.com/dp/B08PCPP729/ref=cm_sw_r_cp_api_glt_i_EXKCNBJADCM0XPR3AVGY?_encoding=UTF8&psc=1

stooged, 2022-01-28:
I wont know until i test it, i have that board coming so i can test when it arrives.

neo305, 2022-01-28:
Ok kool well I can help when I get mine also tomorrow. Last thing. And those 3 H files that are outside the data folder they get flashed even tho they outside or do I put them inside data folder?

closed 2022-01-31 by stooged