stooged/ESP32-Server-900u #10: Not issue, some guidance please

Doyle4, 2022-02-05:
I usually use OS X, Installed Windows as OS X would not detect my ESP32-S2 Mini (little purple one).

Having the same issue.
I have downloaded and installed everything mentioned, as its a clean install of windows, is there any other drivers you would recommended? I don't use Windows so im unsure what to grab sorry.

Once I have tested I can let you know if this board is 100% compatible to add a green tick for others.

stooged, 2022-02-05:
to upload the sketch to the s2 mini you need to put the board into dfu mode

`
Make S2 boards into Device Firmware Upgrade (DFU) mode.
Hold on Button 0
Press Button Reset
Release Button 0 When you hear the prompt tone on usb reconnection
`

https://www.wemos.cc/en/latest/tutorials/s2/get_started_with_arduino_s2.html

neo305, 2022-02-05:
Stooged does that apply to Feathers2 as well? Cuz I just got it and can't get pc to see it. But I think it's cuz I don't have a usb-c data cable. They forgot to mention that was needed lol

Doyle4, 2022-02-05:
> to upload the sketch to the s2 mini you need to put the board into dfu mode
> 
> ```
> 
> Make S2 boards into Device Firmware Upgrade (DFU) mode.
> 
> Hold on Button 0
> 
> Press Button Reset
> 
> Release Button 0 When you hear the prompt tone on usb reconnection
> ```
> 
> https://www.wemos.cc/en/latest/tutorials/s2/get_started_with_arduino_s2.html
> 
> once that is done you upload the sketch with the arduino ide.

Thank you, Will try this asap, thanks again.

Doyle4, 2022-02-05:
[image]

Can confirm ESP32-S2 Mini 100% works with USB emulation!
Thank you to all who helped with this :)

closed 2022-02-05 by Doyle4