stooged/ESP32-Server-900u #11: Flash FeatherS2

neo305, 2022-02-05:
there must be a special way to flash this one cuz I did usb test and it worked but when I flash all i get is failed exploit trigger.

stooged, 2022-02-05:
https://feathers2.io/

```
2. Put your FeatherS2 into download mode by holding [BOOT], clicking [RESET] and then releasing [BOOT]
```

neo305, 2022-02-05:
I have followed echostretch video and can't get this warning off. Maybe that's why?
[image]

neo305, 2022-02-05:
Then after flash I get this warning. Now what do I do? Thanks [image]

stooged, 2022-02-05:
the feather s2 does work, i have one to test.

the error at the end is because the board is in dfu mode so the tool cannot send the reset command, it is normal and you can just hit the reset button.

these 2 i do not get when i upload the code to my feather s2
```
WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
```

is the board you have selected "UM FeatherS2"
this is my board settings for the feather s2
[image]

at the moment you will need to select the 4mb partition because there is no SPIFFS option for a larger partition for some reason.
they have not added it in the library so it will require a manual partition edit to the library or a request on the esp32 github for an update.

neo305, 2022-02-05:
Ok thanks and no I had it set to wrong board lol but still get those library errors. [image]

neo305, 2022-02-05:
Could this be the problem? [image]

stooged, 2022-02-05:
No that is just windows, you have flashed the code to the board.
Now you just plug it into the ps4 and run the exploit.

neo305, 2022-02-06:
Ok how about this. Is it normal after I flash that only Goldhen 2.1.bin is on it? Even tho there are 3 other bin files in the data folder.

neo305, 2022-02-06:
[image]

stooged, 2022-02-06:
goldhen is included in the code that you flashed.

the files in the data folder need to be uploaded to the board through the web interface(File Uploader) or by using the esp sketch data upload tool

stooged, 2022-02-06:
i have created a few tables to enable large spiffs on the 16mb feather board.

https://github.com/stooged/ESP32-Server-900u/tree/main/esplib_mod

closed 2022-02-06 by stooged

scrdcow, 2022-05-26:
Hi! I have the following problem, on Linux Debian Unstable/Sid. I have the latest esptool and pyserial install via pip. Tried two different cables on two different ports. Board is in download mode (BOOT, RST, BOOT). I'm pretty sure this problem is not really related to this project, but I did not find any helpful info online. So I might aswell try to ask :-)

esptool.py v3.3-dev
Serial port /dev/ttyACM0
Traceback (most recent call last):
  File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 5347, in <module>
Connecting...
    _main()
  File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 5340, in _main
    main()
  File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 4647, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 115, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 664, in connect
    last_error = self._connect_attempt(mode=mode, usb_jtag_serial=usb_jtag_serial, extra_delay=extra_delay)
  File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 614, in _connect_attempt
    self.bootloader_reset(usb_jtag_serial, extra_delay)
  File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 592, in bootloader_reset
    self._setDTR(False)  # IO0=HIGH
  File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 532, in _setDTR
    self._port.setDTR(state)
  File "/home/psycho/.local/lib/python3.10/site-packages/serial/serialutil.py", line 603, in setDTR
    self.dtr = value
  File "/home/psycho/.local/lib/python3.10/site-packages/serial/serialutil.py", line 473, in dtr
    self._update_dtr_state()
  File "/home/psycho/.local/lib/python3.10/site-packages/serial/serialposix.py", line 715, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_DTR_str)
OSError: [Errno 71] Protocol error
OSError: [Errno 71] Protocol error

stooged, 2022-05-26:
> Hi! I have the following problem, on Linux Debian Unstable/Sid. UM FeatherS2, the settings is same as years. I have the latest esptool and pyserial installed via pip. Tried two different cables on two different ports. Board is in download mode (BOOT, RST, BOOT). I have no other computer to really test on. I'm sure this problem is not really related to this project since the same problem happnes if I try to erase frash from the terminal with esptool. But I did not find any helpful info online. So I might aswell try to ask :-)
> 
> esptool.py v3.3-dev Serial port /dev/ttyACM0 Traceback (most recent call last): File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 5347, in Connecting... _main() File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 5340, in _main main() File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 4647, in main esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts, File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 115, in get_default_connected_device _esp.connect(before, connect_attempts) File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 664, in connect last_error = self._connect_attempt(mode=mode, usb_jtag_serial=usb_jtag_serial, extra_delay=extra_delay) File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 614, in _connect_attempt self.bootloader_reset(usb_jtag_serial, extra_delay) File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 592, in bootloader_reset self._setDTR(False) # IO0=HIGH File "/home/psycho/.arduino15/packages/esp32/tools/esptool_py/3.3.0/esptool.py", line 532, in _setDTR self._port.setDTR(state) File "/home/psycho/.local/lib/python3.10/site-packages/serial/serialutil.py", line 603, in setDTR self.dtr = value File "/home/psycho/.local/lib/python3.10/site-packages/serial/serialutil.py", line 473, in dtr self._update_dtr_state() File "/home/psycho/.local/lib/python3.10/site-packages/serial/serialposix.py", line 715, in _update_dtr_state fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_DTR_str) OSError: [Errno 71] Protocol error OSError: [Errno 71] Protocol error

It seems it might be an issue with esptool, i only use arduino on a windows machine so i am not much help but when i searched for ``` OSError: [Errno 71] Protocol error ``` a few hits came back with others having the same issue with esptool on non windows environments.

https://github.com/espressif/esptool/issues/333

scrdcow, 2022-05-27:
Oh. thx. I Forgot I could look for issues on esptool on github. Keep up the good work!