stooged/ESP32-Server-900u #12: Esp32 s2 mini

Bladiebloe, 2022-02-05:
This board is working and testen

closed 2022-02-05 by stooged

InfVRN, 2022-02-06:
Hello. Does it wiring free, or need to solder 19, 20 pin?

stooged, 2022-02-06:
> Hello. Does it wiring free, or need to solder 19, 20 pin?

It depends on which board you use, the wiring free boards are listed in the repo.

InfVRN, 2022-02-06:
I told about S2 mini. This board in wiring free in readme, but it doesn't have jumpers or OTG port.

stooged, 2022-02-06:
> I told about S2 mini. This board in wiring free in readme, but it doesn't have jumpers or OTG port.

the s2 mini is wiring free, it has a single usb port for programming and otg

doyle4 has posted confirming the board working here
https://github.com/stooged/ESP32-Server-900u/issues/10#issuecomment-1030695799

InfVRN, 2022-02-06:
Thank you for the answer!