stooged/ESP32-Server-900u #17: Firmware Update, Magic byte wrong.

Doyle4, 2022-02-08:
Is firmware update enabled for S2's? I tried updating via the admin page, I get this error message.
Magic byte is wrong
Expected: 0xE9
Found: 0xFF.
Im wondering if it because the board is not in DFU mode? I put the board into DFU but it loses Wifi connectivity to update.

Thanks.

stooged, 2022-02-08:
DFU mode is for flashing over usb.

OTA is for flashing via the admin page.
if the magic byte is wrong then the file you are trying to update with is corrupt or not a correct update file.

when you flashed the board for the first time if you used a partition scheme that was "no ota" then you cant update via the admin page.

Doyle4, 2022-02-08:
Thank you very much,
Using someone elses .bin to test, I'll check over how I flashed.

Thanks again

stooged, 2022-02-08:
if you have a hex editor open the update file with it and the very first byte should be E9

closed 2022-02-08 by stooged

Doyle4, 2022-02-08:
Hex edit worked, no accepts it.
Only error now is not enough space.

I'll have a word with the person who made the file and see if he will compile without payloads etc.