stooged/ESP32-Server-900u #18: esp32-s2-saola

rajeshca911, 2022-02-11:
**Board** : esp32-s2-saola 
**Soldered** :
[image]
**Issue** : after flashing with arudino IDE, _**no wifi is showing**_ ( flash was successful )

stooged, 2022-02-11:
Does it show the popup "this usb storage devices this filesystem is unsupported" during the exploit process?

rajeshca911, 2022-02-11:
If esp32 library not properly updated then It will throw an error during  compilation/ Upload. So I have left with wiring.  I will re solder with different usb then.
Thanks and I let you knw the update

stooged, 2022-02-11:
> If esp32 library not properly updated then It will throw an error during compilation/ Upload. So I have left with wiring. I will re solder with different usb then. Thanks and I let you knw the update

Not necessarily, the esp32 lib had the tinyusb library ported to it so if it is not the current version it may be missing parts required for msc.

The version i have installed is 2.0.2, it is worth checking that it is up to date.

Once you see the filesystem unsupported message popping up you will know it is all working as intended.

This is the board info for saola

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html

stooged, 2022-02-11:
another way to test the wiring is to flash the USBMSC example to the board and then plug it into a pc, it should show as a usb drive with a txt file on it.

https://github.com/espressif/arduino-esp32/tree/master/libraries/USB/examples/USBMSC

rajeshca911, 2022-02-11:
Bro. It wor6, turned out to be a USB problem.
Its now working perfectly alright.  Thanks alot