stooged/ESP32-Server-900u #20: ESPAsyncWebServer-master.zip

rajeshca911, 2022-02-12:
Arduino: 1.8.19 (Windows 10), Board: "ESP32S2 Dev Module, Disabled, Disabled, Disabled, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

java.io.IOException: **Zip doesn't contain a library**

	

**esp32 Version 2. 0.2**

It worked when first time, but uninstalled yesterday, now I installed again and now It throwing this error when I trying to add library

Mugiwara-v2, 2022-02-12:
no ota spiffs

rajeshca911, 2022-02-12:
Arduino: 1.8.19 (Windows 10), Board: "ESP32S2 Dev Module, Disabled, Disabled, Disabled, UART0, Disabled, No OTA (1MB APP/3MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

java.io.IOException: Zip doesn't contain a library

rajeshca911, 2022-02-13:
Actually,  That zip has been downloaded from github and which is specified in the link ( In your Readme )

My OCD kicking when it used to work before installing but not now. Pls let me knw if you find the problem

stooged, 2022-02-13:
you said the sketch compiled and exported a .bin so why are you still attempting to add a library?

if the sketch compiled and exported the bin file then the espwebserver library is already added.
if the espwebserver was not already installed the sketch would not compile and export the .bin.

your issue revolves around you trying to install a library to the arduino ide that seems to be installed already.