stooged/ESP32-Server-900u #23: How to build bin file to flash

neo305, 2022-02-13:
Sorry not an issue but no where else to ask. Ok so I am able to upload and sketch now. So after sketch I go to temp folder and like I do on the esp8266……what files do I need? And do I build same as I do with esp8266? 
[image]

rajeshca911, 2022-02-13:
Its simple. 
Just 
Step 1. PREPARE  SKETCH
         2. CTRL+ALT+S  ( export compiled binary)
         3. Ctrl  + K        ( sketch folder )
         4.  You see a bin file in the folder

neo305, 2022-02-13:
is there any way to avoid any of these errors? and will it work with all these errors?
Arduino: 1.8.19 (Windows 10), Board: "UM FeatherS2, Enabled, Disabled, Disabled, Enabled, 16M Flash (2MB APP/12.5MB FAT), 240MHz (WiFi), 16MB (128Mb), 921600, None"

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\wifi_drv.cpp: In static member function 'static uint8_t WiFiDrv::getEncTypeNetowrks(uint8_t)':

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\wifi_drv.cpp:451:10: warning: converting to non-pointer type 'uint8_t' {aka 'unsigned char'} from NULL [-Wconversion-null]

   return NULL;

          ^~~~

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\wifi_drv.cpp: In static member function 'static int32_t WiFiDrv::getRSSINetoworks(uint8_t)':

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\wifi_drv.cpp:476:10: warning: converting to non-pointer type 'int' from NULL [-Wconversion-null]

   return NULL;

          ^~~~

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\spi_drv.cpp: In static member function 'static char SpiDrv::spiTransfer(char)':

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\spi_drv.cpp:36:78: warning: integer overflow in expression of type 'long int' results in '-1894967296' [-Woverflow]

 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\spi_drv.cpp:37:26: note: in expansion of macro 'DELAY_SPI'

 #define DELAY_TRANSFER() DELAY_SPI(10)

                          ^~~~~~~~~

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\spi_drv.cpp:76:5: note: in expansion of macro 'DELAY_TRANSFER'

     DELAY_TRANSFER();

     ^~~~~~~~~~~~~~

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\spi_drv.cpp: In static member function 'static void SpiDrv::getParam(uint8_t*)':

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\spi_drv.cpp:36:78: warning: integer overflow in expression of type 'long int' results in '-1894967296' [-Woverflow]

 #define DELAY_SPI(X) { int ii=0; do { asm volatile("nop"); } while (++ii < (X*F_CPU/16000000)); }

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\spi_drv.cpp:37:26: note: in expansion of macro 'DELAY_SPI'

 #define DELAY_TRANSFER() DELAY_SPI(10)

                          ^~~~~~~~~

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\spi_drv.cpp:148:5: note: in expansion of macro 'DELAY_TRANSFER'

     DELAY_TRANSFER();

     ^~~~~~~~~~~~~~

Sketch uses 991606 bytes (47%) of program storage space. Maximum is 2097152 bytes.

Global variables use 46032 bytes (14%) of dynamic memory, leaving 281648 bytes for local variables. Maximum is 327680 bytes.

esptool.py v3.1

Serial port COM7

Connecting...

Chip is ESP32-S2

Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse

Crystal is 40MHz
[... 1843 lines cut ...]
->Writing at 0x00b882d3... (99 %)

->Writing at 0x00d88961... (99 %)

->Writing at 0x00f8c961... (100 %)

->Wrote 12513280 bytes (1040848 compressed) at 0x00411000 in 75.9 seconds (effective 1319.0 kbit/s)...

->Hash of data verified.

->

->Leaving...

->ERROR: ESP32-S2 chip was placed into download mode using GPIO0.

->esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.

->To suppress this error, set --after option to 'no_reset'.

FatFS Upload failed!

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

stooged, 2022-02-13:
You can avoid them by not doing whatever you did to create them.

closed 2022-02-13 by stooged

stooged, 2022-02-13:
You have a lot of issues with this file in the wifi library.

C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\wifi_drv.cpp

neo305, 2022-02-13:
Ok thanks. How do I fix them?

neo305, 2022-02-13:
Do I need to install the library? If so how?
[image]

stooged, 2022-02-13:
You would need to raise an issue on the wifi library github

https://github.com/arduino-libraries/WiFi

neo305, 2022-02-14:
> Its simple. 
> 
> Just 
> 
> Step 1. PREPARE  SKETCH
> 
>          2. CTRL+ALT+S  ( export compiled binary)
> 
>          3. Ctrl  + K        ( sketch folder )
> 
>          4.  You see a bin file in the folder

When you say Prepare Sketch, you mean Upload? Cuz there is u der Sketch Tab Upload. Then there is in Tools Tab Sketch Upload. So which one are you say do? Thanks

stooged, 2022-02-14:
> When you say Prepare Sketch, you mean Upload? Cuz there is u der Sketch Tab Upload. Then there is in Tools Tab Sketch Upload. So which one are you say do? Thanks

if you want to export a .bin file just go to the menu <b>Sketch > Export Compiled Binary</b>
[image]

once it completes it will be in the <b>ESP32_Server_900u</b> folder