stooged/ESP32-Server-900u #24: AI Thinker ESP 12K wiring

linhtranvu, 2022-02-15:
This is the wiring

[image]

This board cannot be powered by connected to 5V and GND. It s supposed to be output as my test

Diagram is here

[image]

If we want to advoid 2 USB connection, more wiring task need to be done by connect red a black of USB data to power source

stooged, 2022-02-15:
i dont have a board to test but this github repo shows that the 5v and gnd pins can be used for otg which was tested by the repo owner.

https://github.com/aziascreations/CircuitPython---AI-Thinker-NODEMCU-32-S2-ESP-12K-08-Target#connecting-to-the-usb-otg-port

with many other boards the 5v rail is connected to the 5v pin but i cant find a schematic for that board so i might just mark it as incompatible because it is a niche product.

closed 2022-02-15 by stooged

linhtranvu, 2022-02-15:
Put aside the wiring for powering board. It works perfectly as a ESP-32-S2 board (could dump code, USB emulation)

THis board is quite popular in SEA countries and China, so it not so rare

So I still suggest to put this board to list to support others user. I will test more with wiring for powering to re-confirm this. And for wiring power without 5V and GND, it is not so hard, just a simple connect red and black to the mini USB input.

linhtranvu, 2022-02-15:
Image of wiring for directly power to 5V and GND, no power up
[image]

stooged, 2022-02-15:
i would say there is a bit of difference between 12k boards because the guy in the other repo used the pins and it worked fine.
the 5v pin should be connected to the usb 5v rail so something is off with the board

i have removed the board from the project list because it does not have enough information available, it has a schematic for the mcu but not the board.

if you look at this listing for it
https://www.tinytronics.nl/shop/en/development-boards/microcontroller-boards/with-wi-fi/ai-thinker-nodemcu-32-s2-esp-12k

it lists the 5v pin as supply voltage which is what it should be
```
Supply voltage:
    Micro USB: 5V DC
    5V pin: 5V DC
    3V3 pin: 3.3V DC
```

it seems problematic so not listing it will keep people from getting it.

EleventhSign, 2022-02-15:
i have the same board and i didn't need to do any special wiring to power the board separately. Server and USB emulation works fine for me with just this:

[image]

[image]
 
there might be an issue with your board @linhtranvu

linhtranvu, 2022-02-16:
With many proof, I doubt I am having problem with wiring, so decide to calculate Voltage and find out my red and black jumper wire from data usb not 5V

It turn out these female wires are oxidized (not used for 3-4 years), so I decide to solder wire directly to board 

Now it works perfectly

So @stooged could you re-edit readme to add this board, and mark as OK. Both I and @EleventhSign have tested it and successfully make it work