stooged/ESP32-Server-900u #27: Cannot upload sketch

subworx, 2022-02-16:
Hi,

Could you please post a description on how exactly you flashed the Wemos S2 mini please? Especially the settings for Arduino IDE.
Also, did you setup any driver in Windows? When booting to DFU mode, I see an unknown device "ESP32-S2" in Device Manager.

The sketch itself compiles fine, all required libraries are there. 
I took all settings from https://www.wemos.cc/en/latest/tutorials/s2/get_started_with_arduino_s2.html ("Serial Connected to: USB CDC" now is named "USB CDC on boot: Enabled", I guess), put the chip into DFU mode as described there, then flashed it without any error messages except that no reset was possible.

Since then it no longer boots normally (after 10 seconds Windows reports a USB device that did not send a device ID, I waited several minutes), DFU mode works still. Tried flashing the sketch and then the data files directly after, didn't help either.

With the PS4-Server-900u, I had absolutely no problems flashing a Wemos D1 mini Pro with the same general software setup.
 Normally I'd claim to be rather computer literate, but now I am absolutely stumped.

Thank you.

stooged, 2022-02-16:
hold the O and while holding it plug the usb plug in.
that will bypass the boot and the board should show up.

subworx, 2022-02-16:
> hold the O button and while holding it plug the usb plug in. that will bypass the boot and the board should show up.
That's the DFU mode I was talking about. I flashed in that mode with the settings listed on the link I pasted.

After flashing, a normal reboot without the O button should work but doesn't.

stooged, 2022-02-16:
you dont really need cdc on boot, you can disable that and just put it into dfu mode when you need to flash.

my s2 mini shows up as unknown in the device manager and it flashes and works fine.

you should be able to just flash the code then plug it into the ps4 and it should work.

stooged, 2022-02-16:
these are my board settings for the s2

https://imgur.com/a/mADOAYQ

stooged, 2022-02-16:
if it does keep playing up put it in dfu mode and use the data upload plugin to wipe the flash and start again

https://github.com/stooged/arduino-esp32fs-plugin

it is the bottom option `!Erase Flash!`

then reboot the board into dfu mode again and flash the sketch again.

subworx, 2022-02-16:
Thank you for the settings screenshot, that helped immensely. Maybe you want to put that into the project somewhere as an orientation for stupids like me?
After another try, I can now see the AP \o/

closed 2022-02-16 by subworx