stooged/ESP32-Server-900u #30: How to sketch and build

neo305, 2022-02-17:
Can someone please tell/show me how to flash/sketch/and build a bin file. Cuz I’m all confused. There is a sketch tab and a tools tab. With esp8266 I would always first under sketch tab upload. 2nd step would be under Tools tab sketch data upload. Is this still the steps for ESP32? Cuz I don’t feel like it’s working correctly. And to then build a bin file for a FeatherS2 I’m clueless cuz anything I try just spits out a 98kb bin file. Please help I’m so frustrated with this.

stooged, 2022-02-17:
compiling and uploading with the arduino ide is the same for every board.
you just load the sketch then select the board, partition scheme, port and then click upload.
when it completes the board has the sketch on it.

you do not need .bin files unless you are uploading it for someone else to use.
if you want to export a .bin file just go to the menu <b>Sketch > Export Compiled Binary</b>
[image]
once it completes it will be in the <b>ESP32_Server_900u</b> folder

echostretch has uploaded a video of him flashing a board 
https://www.youtube.com/watch?v=JMqYdiWf8RI

neo305, 2022-02-17:
ok but are you sure the FeatherS2 doesn't work differently? cuz I have followed echo video and all I get is a 968kb bin file.
so your saying use this
[image]
and not this?
[image]
and thanks btw

stooged, 2022-02-17:
the sketch data upload is to load html and payloads for the exploit, it is not to program the board.

click the arrow in the red circle or upload in the menu to flash the sketch to the board

[image]

stooged, 2022-02-17:
you also have a FAT partition scheme selected so make sure you set this part of the code on line 23

`#define USEFAT false `

to

`#define USEFAT true`

closed 2022-02-18 by stooged