stooged/ESP32-Server-900u #33: Request

LouisGarry, 2022-02-18:
Hi master, if I may request, inthen (internal hen) use goldhen.bin, the reason is for example if there is a new update we can upload it to ESP without reprogramming and the size of goldhen.bin is also smaller.

Thank you.

stooged, 2022-02-18:
set INTHEN to false
https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L27

then upload your goldhen.bin to the board

closed 2022-02-18 by stooged

stooged, 2022-02-18:
You upload the html for autohen in a file called index.html with the goldhen.bin file

```

<!DOCTYPE html>
<html><head>
<meta name="viewport" content="width=device-width, initial-scale=1">
<title>ESP Server</title>
<script>
function setpayload(payload,title,waittime){ 
sessionStorage.setItem('payload', payload); 
sessionStorage.setItem('title', title); 
sessionStorage.setItem('waittime', waittime); 
window.open("loader.html", "_self");}
</script>
<link rel="stylesheet" href="style.css">
</head>
<body onload="setpayload('goldhen.bin','GoldHEN','12000');">
</body>
</html>

```

stooged, 2022-02-18:
I will modify the code so if you upload goldhen.bin it will load that instead of the internally stored payload which will allow updating of the goldhen without flashing

peyman4774, 2022-02-18:
> I will modify the code so if you upload goldhen.bin it will load that instead of the internally stored payload which will allow updating of the goldhen without flashing

need rename gldhen.bin to goldhen.bin in ino file

LouisGarry, 2022-02-18:
Thank you very much master @stooged  and @peyman4774 

I wonder if after goldhen loaded can it auto close the browser?