stooged/ESP32-Server-900u #34: problem running hen

redsam047, 2022-02-18:
sorry for bothering you.
now i successfully upload your original code with no changes to my esp32 and now i get this error every time i run hen
[image]
tried with restarting and power off but no changes same error

stooged, 2022-02-18:
What type of board do you have?

peyman4774, 2022-02-18:
> esp wroom 32

need esp32-s2-wroom

stooged, 2022-02-18:
@redsam047 you need to wire up a usb drive to the board using a circuit like this
https://github.com/stooged/ESP32-Server-900u/blob/main/Images/esp32_diag.jpg

then set 

`#define USBCONTROL false`

to 

`#define USBCONTROL true`

redsam047, 2022-02-18:
i did this before posting 
when i set #define USBCONTROL false everything is ok and get to insert USB manually 
but wen is #define USBCONTROL true i get [this](https://user-images.githubusercontent.com/99916520/154712375-0c51deeb-7a92-4436-a411-94efdbab2b22.jpg)

> @redsam047 you need to wire up a usb drive to the board using a circuit like this https://github.com/stooged/ESP32-Server-900u/blob/main/Images/esp32_diag.jpg
> 
> then set
> 
> `#define USBCONTROL false`
> 
> to
> 
> `#define USBCONTROL true`

peyman4774, 2022-02-18:
> and another question do i need change USBpin 4 to usbPin 5 because in my bord gpio4 is pin number 5?

change gpio in ino file

stooged, 2022-02-18:
@redsam047 you need to look up the pinout schematic for your board and the pin you have the wire connected to look for GPIO and what ever that number is you set the usbPin to.

if it is GPIO4 then it is usbPin 4
if it is GPIO7 then it is usbPin 7

closed 2022-02-19 by stooged

redsam047, 2022-02-19:
@[stooged](https://github.com/stooged) 
the same problem happens again
this is my bord[ pinout](https://www.electronicshub.org/wp-content/uploads/2021/02/ESP32-Pinout-1.jpg)
i did solder them this way(based on [this](https://github.com/stooged/ESP32-Server-900u/blob/main/Images/esp32_diag.jpg?raw=true))
[image]
[image]