stooged/ESP32-Server-900u #35: Help

EdwardDanielSalazar, 2022-02-18:
hello good day, sorry for the inconvenience, is there a way to convert goldhen.bin to goldhen.h or any payload, so that when a new hen comes out, include it myself?, thanks in advance

peyman4774, 2022-02-18:
> hello good day, sorry for the inconvenience, is there a way to convert goldhen.bin to goldhen.h or any payload, so that when a new hen comes out, include it myself?, thanks in advance

update the sketche code & upload goldhen.bin to spiffs

keenware, 2022-02-19:
> > hello good day, sorry for the inconvenience, is there a way to convert goldhen.bin to goldhen.h or any payload, so that when a new hen comes out, include it myself?, thanks in advance
> 
> update the sketche code & upload goldhen.bin to spiffs

Can u explain further, maybe the step to do it. Thanks

stooged, 2022-02-19:
@EdwardDanielSalazar  added a tool to generate the goldhen.h file from the goldhen.bin file

https://github.com/stooged/ESP32-Server-900u/tree/main/bin2h

closed 2022-02-19 by stooged