stooged/ESP32-Server-900u #36: ESP32-S2-SAOLA WIFI

rajeshca911, 2022-02-19:
after uploading sketch to the board, connected wifi with default password.
went to 10.1.1.1\admin.html =>config =>changed Home_wifi to My wifi SSID and password to Mine. later unchecked AP check box.
checked connect to wifi box. => config saved => reboot

but problem is esp 32 s2 not showing in the wlan clients list
where it went wrong ?

stooged, 2022-02-19:
are there any special characters in the password?
like !@#$%^&*()

i have not tested it with special characters

rajeshca911, 2022-02-19:
yes @ is there

stooged, 2022-02-19:
ok i will have a look and see if the chars are being converted

stooged, 2022-02-19:
i looked at the password and it is saving properly so all i can suggest is to use the esp sketch data upload tool to wipe the board and upload the flash and try again.

when you do that leave the ap running until you know the wifi is connecting and then disable the ap

rajeshca911, 2022-02-19:
Nope not worked.
changed Home_Wifi password to normal characters yet results were unchanged. guess I have to stick it to the offline or it may be esp32-s2-saola problem

rajeshca911, 2022-02-19:
is there any password char lenth limit ?

stooged, 2022-02-19:
i dont think it will be the board 

you could test it using a wifi example sketch from the esp32 library

https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/SimpleWiFiServer

closed 2022-02-20 by stooged