stooged/ESP32-Server-900u #37: Problem Uploading Files on File Manager

keenware, 2022-02-19:
U have an suggestion for error upload file.bin using the sketch data upload (success but file not appear in file manager). I use file upload its not appear in file manager too.

keenware, 2022-02-19:
[image]
[image]
[image]
[image]

stooged, 2022-02-19:
go to the "ESP Information" and copy paste the `###### Storage information ######`  here

keenware, 2022-02-19:
> go to the "ESP Information" and copy paste the `###### Storage information ######` here
###### Software ######

Firmware version 1.00
SDK version: v4.4-beta1-189-ga79dc75f0a
###### Board ######

MCU: ESP32S2
Chip Id: ESP32-S2
CPU frequency: 240MHz
Cores: 1
###### Flash chip information ######

Flash chip Id: 255
Estimated Flash size: 0 B
Flash frequency: 0.00 MHz
Flash write mode: UNKNOWN
###### Storage information ######

Filesystem: SPIFFS
Total Size: 0 B
Used Space: 0 B
Free Space: 0 B
###### PSRam information ######

Psram Size: 0 B
Free psram: 0 B
Max alloc psram: 0 B
###### Ram information ######

Ram size: 193.36 KB
Free ram: 105.25 KB
Max alloc ram: 87.99 KB
###### Sketch information ######

Sketch hash: d41d8cd98f00b204e9800998ecf8427e
Sketch size: 0 B
Free space available: 1.25 MB

stooged, 2022-02-19:
Filesystem: SPIFFS
Total Size: 0 B
Used Space: 0 B
Free Space: 0 B

you have a problem with the selected partition on the board.
there is no storage so you cant upload.

make sure you have a spiffs partition selected or set `USEFAT false` to `USEFAT true` if you are using a fat partition.

closed 2022-02-19 by stooged