stooged/ESP32-Server-900u #38: Files Not Showing Up In File Manager

EchoStretch, 2022-02-19:
When having lots of files as in trainer files the page doesn't load. maybe add a max for 25 then page 1 page 2 option?

oxstretchxo, 2022-02-20:
Also maybe download all button.

stooged, 2022-02-20:
it should be fixed now, i tested it with 170 files listed and it did not exceed memory required. 

it is tricky to "download all" with the esp board because there is a hard limit on the number of files that can be opened at a single time and memory constraints.

closed 2022-02-20 by stooged

EchoStretch, 2022-02-20:
when i use trainers its 1k+ files but ill give it a try

EchoStretch, 2022-02-20:
tested and fixed

stooged, 2022-02-21:
@oxstretchxo i have added a download all function to the file manager, it may or may not work very well.

oxstretchxo, 2022-02-21:
Ok ill give it a go and give you an update