stooged/ESP32-Server-900u #40: wifi on esp32

redsam047, 2022-02-20:
first of all thanks to @[stooged](https://github.com/stooged/ESP32-Server-900u/commits?author=stooged) for your help
i successfully setup my esp32 with USB working now i have a new problem.
when i set my wifi SSID,esp does not connect to my wifi.
 i try this [wifi example](https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/SimpleWiFiServer) sketch from the esp32 library and on this one esp connect to wifi scssesfully

Doyle4, 2022-02-20:
Make sure your wifi is 2.4ghz and not 5ghz, these boards do not have 5ghz wifi.

stooged, 2022-02-20:
Do you have spaces in the pw or ssid?

neo305, 2022-02-20:
For the record I always set mine up with the password "" (I leave it blank) and I've never had a problem with connection.

redsam047, 2022-02-20:
> Do you have spaces in the pw or ssid?

no.i even changed password to 12345678 and still does not connect to my home wifi

neo305, 2022-02-20:
> > Do you have spaces in the pw or ssid?
> 
> 
> 
> no.i even changed password to 12345678 and still does not connect to my home wifi

You check your firewall?

redsam047, 2022-02-20:
> > > Do you have spaces in the pw or ssid?
> > 
> > 
> > no.i even changed password to 12345678 and still does not connect to my home wifi
> 
> You check your firewall?

its disabled
i don't think the problem is from my wifi because when use [wifi example](https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/SimpleWiFiServer) esp conect to it and obtain ip

stooged, 2022-02-20:
i tried a number of tests on the pw and ssid to see if it is passing the information properly which it is.
the board connects in my tests to my mesh network.

so something is happening that i cant diagnose, all i can suggest is to hard code your ssid and pw

https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L685

change 

` WiFi.begin(WIFI_SSID.c_str(), WIFI_PASS.c_str()); `

to 

' WiFi.begin( "your_wifi_ssid", "your_password" );'

then upload the sketch and see if it works.

redsam047, 2022-02-20:
i try that but still doesn't connect to wifi.
try with another router but still no result

Doyle4, 2022-02-20:
Try:
Enter the wifi’s ssid you are wanting to connect the board to and the password for that ssid.

save and it will reboot.
Power off the ESP32-S2 after reboot.
Power off PS4
Reboot your router, once rebooted, boot PS4, then power the esp32-s2 and leave it a good 5mins. 
then see. I had issues and I remember this working for me and has stayed connected since.

also make sure the wifi box is ticked in the config section.

stooged, 2022-02-20:
> i try that but still doesn't connect to wifi. try with another router but still no result

Are you ticking the checkbox to enable the wifi connection?

stooged, 2022-02-20:
i had a thought that some people might be setting the wifi settings in the default settings part of the code and expecting those settings to apply after they had already run the sketch and generated a config file.

those settings were originally intended to be set through the config webpage so i have added an option to disable the config file and permanently use the default settings listed in the sketch.

if you want to disable the config.ini then change

`#define USECONFIG true `

to 

`#define USECONFIG false `

 then set your wifi info / settings in the sketch and upload it to the board

closed 2022-02-20 by stooged

neo305, 2022-02-20:
Yeah I'm gonna need a how to video on this lol

Doyle4, 2022-02-21:
Thanks for the update, this has resolved an issue that was bugging me, my mac would always detect the default ssid even when changed, I thought it was my mac playing up, but this has solved it for me thanks.

redsam047, 2022-02-21:
> > i try that but still doesn't connect to wifi. try with another router but still no result
> 
> Are you ticking the checkbox to enable the wifi connection?

yes

redsam047, 2022-02-21:
0

> those settings were originally intended to be set through the config webp

yes

Doyle4, 2022-02-21:
> Yeah I'm gonna need a how to video on this lol

This isn't exactly what you need, but a very good starting point, watch both videos (part 1 and part 2), once watched you will have a much better understanding what is been said in this post :)

https://www.youtube.com/watch?v=g_EiQt5DYdQ
https://www.youtube.com/watch?v=o_OT8gMChTs&t=9s

Please dont jump to part 2 as its for connecting to wifi, make sure to start from part 1 as it includes building the firmware etc.

I hope this helps.