stooged/ESP32-Server-900u #41: Maby a working board

Bladiebloe, 2022-02-20:
Hey stooged maby this board is also good ????
wemos-lite-v1-esp32-wifi-bluetooth

Mugiwara-v2, 2022-02-20:
not s2 lol

Bladiebloe, 2022-02-20:
Package included: 1 x WEMOS LOLIN32 Lite V1.0.0 Wifi and bluetooth Board Based 2 x 13P pinhead

WIFI bluetooth 4MB Flash Lastest ESP32 Version: REV1 Max serial baud rate: 256000bps

Features: Lithium battery interface, 500mA Max charging current Default firmware: Lastest MicroPython firmware

Bladiebloe, 2022-02-20:
It have a 4mb flash

stooged, 2022-02-20:
that is a esp32 board.
all esp32 boards work but you must wire a usb drive to the board or manually plug/unplug a usb drive

https://github.com/stooged/ESP32-Server-900u/blob/main/Images/esp32_diag.jpg

closed 2022-02-20 by stooged