stooged/ESP32-Server-900u #43: Update

EchoStretch, 2022-02-23:
Just did a fresh install of windows 10 and was unable to use
ESP32-S2-DevKitC-1
ESP32-S2-DevKitM-1

without updating drivers
https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers

Not sure if u wanna add that in the list.

Feather S2 worked fine

stooged, 2022-02-24:
the arduino ide installer should install the usb to serial/uart drivers during the install process

[image]

i tested it on a fresh network isolated win10 vm with my devkit-m board and it worked fine.
did you install the arduino ide from the zip file because that does require manual installation of the drivers.

i will add a note for drivers in the readme just incase there is a version of the installer floating around without the drivers.

closed 2022-02-24 by stooged

oxstretchxo, 2022-02-24:
I installed from windows store. On windows 10

stooged, 2022-02-24:
ahh i see, the store version must not include the driver package.
i have always use the installer from the site.

i added a note about the drivers anyway, thanks for the heads up.