stooged/ESP32-Server-900u #45: Uploading Files With FatFs

EchoStretch, 2022-03-03:
only got about 3mb of files using feathers2 but its just keeps saying error failed to open

[image]

_>error: failed to open "/spiflash/CUSA11229_01.03.json.gz" for writing
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11229_01.03.json.gzdir_d: /spiflash/CUSA11229_01.03.json.gz  file name: CUSA11229_01.03.json.gz
_>error: failed to open "/spiflash/CUSA11253.jpg.gz" for writing
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11253.jpg.gzdir_d: /spiflash/CUSA11253.jpg.gz  file name: CUSA11253.jpg.gz
_>error: failed to open "/spiflash/CUSA11253_01.17.json.gz" for writing
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11253_01.17.json.gzdir_d: /spiflash/CUSA11253_01.17.json.gz  file name: CUSA11253_01.17.json.gz
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11260.jpg.gzdir_d: /spiflash/CUSA11260.jpg.gz  file name: CUSA11260.jpg.gz
_>error: failed to open "/spiflash/CUSA11260.jpg.gz" for writing
_>error: failed to open C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11260_01.12.json.gz for reading
_>error adding file!
_>
->g_flashmem[0]: �size: 12513280

FatFS Create Failed!

stooged, 2022-03-03:
`error: failed to open /spiflash/`

it appears to be trying to open a directory on the board that does not exist

are you using the upload plugin i modified?
https://github.com/stooged/arduino-esp32fs-plugin

during the upload process make sure you select "FatFS" in the popup box.

closed 2022-03-04 by stooged

EchoStretch, 2022-03-04:
C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master>cd C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\\src

C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\src>set PATH=C:\Program Files\Java\jdk-17.0.2\bin;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Users\EchoStretch\AppData\Local\Microsoft\WindowsApps;

C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\src>set ALP=C:\Program Files (x86)\Arduino\lib

C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\src>IF EXIST bin GOTO NODIR

C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\src>del bin\*.jar
Could Not Find C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\src\bin\*.jar

C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\src>rd /S /Q bin\com
The system cannot find the file specified.

C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\src>javac.exe -target 1.8 -cp ".;C:\Program Files (x86)\Arduino\lib\arduino-core.jar;C:\Program Files (x86)\Arduino\lib\commons-codec-1.7.jar;C:\Program Files (x86)\Arduino\lib\pde.jar" -d bin ESP32FS.java
warning: target release 1.8 conflicts with default source release 17

C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\src>cd bin

C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\src\bin>jar.exe cvfM esp32fs.jar *
adding: ESP32FS.java(in = 23042) (out= 5308)(deflated 76%)

C:\Users\EchoStretch\Desktop\arduino-esp32fs-plugin-master\src\bin>pause
Press any key to continue . . .

===========================================================

had to move the ESP32FS.java into the bin folder but when its done and i try using esp32fs.jar it doesnt come up in arduino
C:\Program Files (x86)\Arduino\tools\ESP32FS\tool\esp32fs.jar

stooged, 2022-03-04:
the file and folder structure does not need to be changed and the ESP32FS.java should be in the src folder

the source targets 1.8 and you are getting this warning
`warning: target release 1.8 conflicts with default source release 17` so this might be causing issues.

i installed jdk 1.8.0_202 and used that to compile the tool.

on windows the compiled esp32fs.jar will be in the bin folder

you move that to 'C:\Users\EchoStretch\Documents\Arduino\tools\ESP32FS\tool\'

you will also need the mkfatfs binary for fatfs
https://github.com/labplus-cn/mkfatfs

<hr>

the easiest way is to just use the compiled release i provided 
https://github.com/stooged/arduino-esp32fs-plugin/releases/download/2.0.7/ESP32FS.zip

and extract it into `C:\Users\EchoStretch\Documents\Arduino\tools\`

EchoStretch, 2022-03-04:
ok so i compiled it and it showed up but i get the same error and used yours too same error what is spiflash? and why do i see it?

_>error: failed to open "/spiflash/CUSA11112.jpg.gz" for writing
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11112.jpg.gzdir_d: /spiflash/CUSA11112.jpg.gz  file name: CUSA11112.jpg.gz
_>error: failed to open "/spiflash/CUSA11112_01.01.json.gz" for writing
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11112_01.01.json.gzdir_d: /spiflash/CUSA11112_01.01.json.gz  file name: CUSA11112_01.01.json.gz
_>error: failed to open "/spiflash/CUSA11115.jpg.gz" for writing
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11115.jpg.gzdir_d: /spiflash/CUSA11115.jpg.gz  file name: CUSA11115.jpg.gz
_>error: failed to open "/spiflash/CUSA11115_01.00.json.gz" for writing
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11115_01.00.json.gzdir_d: /spiflash/CUSA11115_01.00.json.gz  file name: CUSA11115_01.00.json.gz
_>error: failed to open "/spiflash/CUSA11229.jpg.gz" for writing
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11229.jpg.gzdir_d: /spiflash/CUSA11229.jpg.gz  file name: CUSA11229.jpg.gz
_>error: failed to open "/spiflash/CUSA11229_01.03.json.gz" for writing
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11229_01.03.json.gzdir_d: /spiflash/CUSA11229_01.03.json.gz  file name: CUSA11229_01.03.json.gz
_>error: failed to open "/spiflash/CUSA11253.jpg.gz" for writing
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11253.jpg.gzdir_d: /spiflash/CUSA11253.jpg.gz  file name: CUSA11253.jpg.gz
_>error: failed to open "/spiflash/CUSA11253_01.17.json.gz" for writing
_>error: failed to open C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11260.jpg.gz for reading
_>error adding file!
_>
->Pack file, dir_s: C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11253_01.17.json.gzdir_d: /spiflash/CUSA11253_01.17.json.gz  file name: CUSA11253_01.17.json.gz
->g_flashmem[0]: �size: 12513280

FatFS Create Failed!

EchoStretch, 2022-03-04:
ok fixed that issue when i changed to your mkfatfs.. so now i get this..

_>error: failed to open "/fatfs/CUSA11115_01.00.json.gz" for writing
_>error: failed to open "/fatfs/CUSA11229.jpg.gz" for writing
_>error: failed to open "/fatfs/CUSA11229_01.03.json.gz" for writing
_>error: failed to open "/fatfs/CUSA11253.jpg.gz" for writing
_>error: failed to open "/fatfs/CUSA11253_01.17.json.gz" for writing
_>error: failed to open "/fatfs/CUSA11260.jpg.gz" for writing
_>error: failed to open C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11260_01.12.json.gz for reading
_>error adding file!
_>
_>error: failed to open image file

FatFS Create Failed!

stooged, 2022-03-04:
that looks better as it is looking for the fatfs folder.

go into the admin panel of the board and copy paste the `###### Storage information ######` section here

EchoStretch, 2022-03-04:
###### Software ######

Firmware version 1.00
SDK version: v4.4-beta1-189-ga79dc75f0a
###### Board ######

MCU: ESP32S2
Board: FeatherS2
Chip Id: ESP32-S2
CPU frequency: 240MHz
Cores: 1
###### Flash chip information ######

Flash chip Id: 2
Estimated Flash size: 16.00 MB
Flash frequency: 80.00 MHz
Flash write mode: DIO
###### Storage information ######

Filesystem: FatFs
Total Size: 11.81 MB
Used Space: 4.00 KB
Free Space: 11.80 MB
###### PSRam information ######

Psram Size: 8.00 MB
Free psram: 7.94 MB
Max alloc psram: 7.87 MB
###### Ram information ######

Ram size: 191.90 KB
Free ram: 105.41 KB
Max alloc ram: 87.99 KB
###### Sketch information ######

Sketch hash: e7ce8e0041926dda8ad6332fce349869
Sketch size: 1003.47 KB
Free space available: 1.02 MB

stooged, 2022-03-04:
The fatfs partition is there so i dont know why it is not uploading.

I use the same esp32fs tool and it uploads to my feathers2 board with no problems on win11.

EchoStretch, 2022-03-04:
yea i have no idea even swtiched to the setup and not the windows store.. maybe windows 10 issue but i cant see it.

EchoStretch, 2022-03-04:
i loaded a few html files and works fine... maybe gz files?

EchoStretch, 2022-03-04:
no i extracted everything. samething
_>error: failed to open "/fatfs/CUSA11941.jpg" for writing
_>error: failed to open "/fatfs/CUSA11941_01.02.json" for writing
_>error: failed to open C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u\data\CUSA11993.jpg for reading
_>error adding file!
_>
_>error: failed to open image file

FatFS Create Failed!

EchoStretch, 2022-03-04:
might have it.. I dont think it likes 1000 files maybe tomany

EchoStretch, 2022-03-05:
ok just an update.. Strange update....

700 files I get error but 600 files is good and finishes.. Never tried in between.. 

->Writing at 0x0057da71... (97 %)
->Writing at 0x00733bc7... (97 %)
->Writing at 0x00937bc7... (98 %)
->Writing at 0x00b38963... (98 %)
->Writing at 0x00d3c963... (99 %)
->Writing at 0x00f3d1f5... (100 %)
->Wrote 12513280 bytes (346891 compressed) at 0x00411000 in 81.0 seconds (effective 1235.8 kbit/s)...
->Hash of data verified.
->
->Leaving...

But only 171 files get uploaded and even when I try to download all it only gives me the 171 files.

stooged, 2022-03-05:
it is a problem with mkfatfs

in my tests i cant get it to generate the image with more than 201 files

you can test it without a board using the following command
,,,
mkfatfs.exe -c Data -s 0xbef000 -t fatfs test.bin
,,,

EchoStretch, 2022-03-05:
yep its the mkfatfs.exe error when using that command with 1075 files

error: failed to open "/fatfs/CUSA11229_01.03.json.gz" for writing
error: failed to open "/fatfs/CUSA11253.jpg.gz" for writing
error: failed to open "/fatfs/CUSA11253_01.17.json.gz" for writing
error: failed to open "/fatfs/CUSA11260.jpg.gz" for writing
error: failed to open Data\CUSA11260_01.12.json.gz for reading
error adding file!

error: failed to open image file

C:\Users\EchoStretch\Desktop\ESP32-Server-900u-main\ESP32_Server_900u>

stooged, 2022-03-05:
it does not seem to be size related because i can do 201 larger files so it is a file count issue.

EchoStretch, 2022-03-05:
yea im doing 1k+ at small sizes
[image]

stooged, 2022-03-05:
it seems to be an issue with the tools fatfs, it works with 1000+ for me using littlefs

```
mkfatfs.exe -c Data -s 0xbef000 -t littlefs test.bin
```

EchoStretch, 2022-03-05:
yep that worked so how do we change the partition to littlefs.. lol

stooged, 2022-03-05:
for now you should be able to use spiffs but you will need to add custom spiffs partitions to the esp lib

they are here
https://github.com/stooged/ESP32-Server-900u/tree/42c96a27af2a55748eeeb86c58669256c62708db/esplib_mod

once you add those files you can select a 16M Flash (SPIFFS) partition and then use the upload tool with SPIFFS selected.

make sure you set `#define USEFAT false` 

i tested it with 1013 of the json/jpg files from the trainer and they all uploaded

stooged, 2022-03-05:
doing more testing and it turns out it is a limitation with fatfs.
there is a limit of 512 entries per directory and any file that is not in the 8:3 format uses 2 or 3 entries.

i tested the tool by splitting the 1013 files into 7 directories each containing about 150 files and it created the image.

oxstretchxo, 2022-03-05:
Interesting but fatfs is better then spiff? Correct for stability?

stooged, 2022-03-05:
> Interesting but fatfs is better then spiff? Correct for stability?

i am not sure

spiffs has been fine on the esp8266 boards for a long time even with large partitions for the d1 mini pro 16mb boards.
spiffs is slower for read and write when compared to fatfs but for the ps4 webhost it should not matter too much.

you could still use fatfs but you would need to use multiple folders and code your trainer to load from all the folders.

EchoStretch, 2022-03-06:
this is coming along great.. so much space only problem left is joining the bins.. using correct offsets and has no issue doing it with the 4mb with the offsets for 4mb but 16mb i cant get it. have you tried do combine a 16mb bin?

EchoStretch, 2022-03-06:
ok so im trying to fix this error
->
->Leaving...
->ERROR: ESP32-S2 chip was placed into download mode using GPIO0.
->esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
->To suppress this error, set --after option to 'no_reset'.
SPIFFS Upload failed!

when sketching its located in platforms
tools.esptool_py.upload.pattern_args=--chip {build.mcu} --port "{serial.port}" --baud {upload.speed} {upload.flags} --before default_reset --after no_reset write_flash -z --flash_mode {build.flash_mode} --flash_freq {build.flash_freq} --flash_size {build.flash_size} 0xe000 "{runtime.platform.path}/tools/partitions/boot_app0.bin" {build.bootloader_addr} "{build.path}/{build.project_name}.bootloader.bin" 0x10000 "{build.path}/{build.project_name}.bin" 0x8000 "{build.path}/{build.project_name}.partitions.bin" {upload.extra_flags}

but when data upload im not sure maybe the mkfatfs.jar

stooged, 2022-03-06:
Leaving...
->ERROR: ESP32-S2 chip was placed into download mode using GPIO0.
->esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
->To suppress this error, set --after option to 'no_reset'.
SPIFFS Upload failed!

That error is because the esptool cant send the exit command to the board.
When you use the button to enter the flash mode it requires a hard reset at that point which is pressing the rst button.

To get around it while testing things just enable cdc on boot in the board settings and flash the sketch.
It should then switch com port and you can then flash the board without needing to press the button to put it into flash mode.

EchoStretch, 2022-03-06:
yea usb cdc on boot is enabled.. thats what you mean correct?

stooged, 2022-03-06:
Yes.
You should be able to flash the board without pressing the buttons.

EchoStretch, 2022-03-06:
yea i tried everything to combine these bins with flash download tool 3.9.2 and everytime i flash with NodeMCU-PyFlasher it doesnt come up with wifi

stooged, 2022-03-06:
I cant help with that as i dont combine bins or use pyflasher.

I only use the arduino ide to flash the board.