stooged/ESP32-Server-900u #46: fatfs setting

Mugiwara-v2, 2022-03-05:
not good  css not detect includ css in html works fine

Mugiwara-v2, 2022-03-05:
static const uint8_t style_gz[] PROGMEM = {
    31, 139, 8, 8, 92, 31, 20, 98, 0, 11, 115, 116, 121, 108, 101, 46, 99, 115, 115, 0, 189, 86, 221, 106, 219, 48, 20, 190, 15, 228, 29, 68,
    67, 97, 129, 186, 56, 105, 26, 90, 155, 66, 55, 218, 23, 216, 237, 216, 133, 108, 201, 142, 136, 44, 25, 73, 110, 146, 150, 190, 251, 142, 100, 203,
    118, 18, 183, 205, 74, 55, 27, 130, 115, 116, 254, 244, 157, 243, 29, 41, 145, 100, 135, 94, 198, 163, 4, 167, 235, 92, 201, 74, 144, 32, 149, 92,
    170, 8, 77, 102, 139, 235, 217, 247, 199, 120, 60, 242, 130, 204, 61, 32, 200, 164, 48, 129, 102, 207, 52, 66, 179, 69, 185, 245, 146, 13, 101, 249,
    202, 68, 40, 145, 156, 128, 172, 192, 42, 103, 34, 66, 97, 253, 94, 134, 32, 147, 79, 84, 101, 92, 110, 130, 93, 180, 98, 132, 80, 1, 178, 18,
    19, 194, 68, 14, 138, 151, 11, 90, 28, 252, 46, 105, 1, 42, 175, 104, 60, 26, 143, 46, 19, 35, 108, 174, 70, 97, 161, 153, 97, 82, 4, 164,
    82, 216, 126, 56, 99, 13, 154, 137, 220, 6, 122, 133, 137, 220, 216, 192, 55, 229, 22, 205, 150, 240, 19, 34, 149, 39, 248, 91, 120, 225, 222, 203,
    249, 244, 2, 68, 118, 97, 30, 30, 175, 206, 110, 167, 241, 16, 34, 15, 146, 228, 84, 253, 224, 21, 117, 129, 20, 161, 32, 20, 82, 208, 14, 163,
    205, 138, 25, 218, 223, 211, 108, 222, 164, 112, 0, 91, 45, 73, 43, 165, 173, 89, 41, 153, 48, 84, 189, 1, 228, 171, 223, 124, 180, 178, 248, 13,
    151, 235, 167, 220, 97, 222, 228, 86, 27, 104, 158, 26, 244, 255, 240, 50, 116, 107, 2, 204, 89, 14, 206, 83, 218, 108, 7, 182, 146, 172, 25, 200,
    203, 146, 98, 72, 35, 165, 45, 98, 95, 131, 239, 173, 205, 248, 11, 208, 181, 96, 253, 29, 188, 5, 102, 174, 27, 235, 54, 15, 56, 205, 192, 231,
    236, 26, 240, 137, 145, 125, 250, 77, 96, 65, 155, 185, 21, 144, 202, 186, 22, 17, 194, 137, 150, 188, 130, 253, 128, 212, 200, 18, 10, 96, 191, 84,
    157, 93, 232, 48, 48, 70, 22, 141, 188, 14, 240, 22, 137, 234, 156, 74, 238, 179, 234, 72, 101, 235, 229, 224, 248, 135, 129, 113, 196, 153, 88, 219,
    184, 7, 179, 2, 53, 125, 65, 104, 42, 125, 231, 53, 21, 173, 57, 141, 163, 39, 6, 105, 81, 242, 57, 227, 182, 98, 39, 152, 66, 53, 169, 130,
    60, 251, 246, 56, 53, 236, 137, 126, 218, 129, 193, 9, 119, 214, 174, 177, 50, 92, 48, 190, 3, 120, 21, 195, 252, 2, 105, 96, 93, 160, 169, 98,
    206, 91, 221, 207, 182, 157, 56, 46, 53, 52, 169, 255, 178, 139, 27, 70, 204, 202, 246, 73, 120, 222, 249, 38, 23, 200, 172, 92, 55, 54, 84, 128,
    62, 71, 80, 56, 70, 208, 132, 184, 39, 70, 125, 210, 217, 58, 197, 168, 45, 252, 141, 43, 122, 237, 139, 137, 178, 50, 191, 204, 174, 164, 119, 103,
    186, 74, 10, 102, 206, 126, 123, 236, 80, 191, 221, 251, 83, 190, 65, 36, 87, 20, 10, 141, 154, 74, 15, 122, 242, 40, 130, 146, 172, 140, 69, 40,
    216, 51, 62, 240, 53, 28, 208, 71, 152, 104, 202, 51, 102, 97, 181, 28, 58, 100, 117, 43, 243, 44, 22, 82, 21, 152, 199, 3, 233, 37, 21, 52,
    177, 232, 54, 250, 193, 62, 39, 161, 123, 222, 245, 212, 181, 203, 193, 62, 59, 227, 35, 193, 112, 208, 102, 234, 48, 32, 17, 126, 178, 14, 125, 11,
    44, 14, 201, 154, 177, 45, 181, 115, 234, 57, 96, 208, 127, 91, 80, 137, 27, 214, 206, 107, 213, 102, 242, 212, 127, 246, 162, 45, 23, 183, 215, 143,
    15, 125, 226, 130, 249, 241, 185, 107, 167, 126, 120, 148, 19, 222, 27, 35, 118, 246, 91, 189, 101, 119, 156, 189, 197, 205, 227, 219, 193, 209, 5, 130,
    48, 93, 114, 12, 84, 73, 184, 76, 215, 199, 161, 143, 121, 221, 94, 70, 156, 102, 161, 243, 97, 206, 238, 181, 198, 120, 212, 52, 135, 39, 179, 63,
    232, 252, 16, 113, 237, 198, 37, 38, 142, 7, 104, 15, 99, 212, 209, 178, 153, 233, 227, 209, 170, 113, 220, 10, 14, 174, 56, 78, 230, 217, 186, 236,
    216, 154, 93, 217, 183, 183, 26, 40, 76, 88, 165, 173, 167, 243, 190, 216, 149, 181, 103, 120, 181, 184, 189, 33, 201, 105, 185, 180, 71, 173, 96, 69,
    83, 16, 93, 194, 97, 48, 215, 200, 118, 42, 86, 136, 137, 140, 9, 123, 126, 90, 245, 19, 212, 28, 66, 86, 247, 222, 251, 94, 211, 93, 166, 112,
    65, 117, 109, 3, 37, 8, 207, 95, 144, 95, 117, 247, 140, 12, 32, 143, 144, 146, 6, 27, 250, 45, 36, 52, 159, 198, 8, 220, 216, 209, 134, 222,
    83, 189, 90, 118, 202, 174, 48, 247, 67, 193, 192, 197, 199, 81, 78, 112, 62, 209, 32, 175, 116, 59, 186, 247, 238, 13, 239, 204, 151, 9, 116, 94,
    86, 152, 33, 187, 119, 39, 211, 244, 238, 238, 44, 254, 3, 92, 200, 253, 182, 108, 11, 0, 0};
#endif

Mugiwara-v2, 2022-03-05:
not works in fat fs

stooged, 2022-03-07:
there was a code issue in that stored style.css, i have uploaded the a fix for the error.

closed 2022-03-07 by stooged