stooged/ESP32-Server-900u #48: Move to LittleFS for better performance?

OpenGG, 2022-03-09:
From my test result, LittleFS has better average performance than SPIFFS on ESP32S2.

Three cases were tested with wrk:
1. 404-notfound
2. LICENSE: MIT LICENSE file, around 1k
3. merged-flash.bin: arduino binary file, around 1MB

Test cli: `wrk -t6 -c6 -d30s http://10.1.1.1/404-notfound && wrk -t6 -c6 -d30s http://10.1.1.1/LICENSE && wrk -t1 -c1 -d30s http://10.1.1.1/merged-flash.bin`

LittleFS:

```
Running 30s test @ http://10.1.1.1/404-notfound
  6 threads and 6 connections
  Thread Stats   Avg      Stdev     Max   +/- Stdev
    Latency    46.20ms   26.03ms 372.25ms   89.03%
    Req/Sec    18.15      7.99    50.00     55.66%
  2385 requests in 30.10s, 279.49KB read
  Socket errors: connect 0, read 13, write 0, timeout 0
  Non-2xx or 3xx responses: 2385
Requests/sec:     79.23
Transfer/sec:      9.28KB
Running 30s test @ http://10.1.1.1/LICENSE
  6 threads and 6 connections
  Thread Stats   Avg      Stdev     Max   +/- Stdev
    Latency    51.30ms   14.44ms  96.26ms   64.77%
    Req/Sec    16.20      6.31    40.00     71.95%
  2325 requests in 30.09s, 2.70MB read
  Socket errors: connect 0, read 12, write 0, timeout 0
Requests/sec:     77.26
Transfer/sec:     91.98KB
Running 30s test @ http://10.1.1.1/merged-flash.bin
  1 threads and 1 connections
  Thread Stats   Avg      Stdev     Max   +/- Stdev
    Latency     1.89s    91.78ms   1.97s    85.71%
    Req/Sec     0.00      0.00     0.00    100.00%
  15 requests in 30.04s, 15.79MB read
  Socket errors: connect 0, read 0, write 0, timeout 8
Requests/sec:      0.50
Transfer/sec:    538.43KB
```

SPIFFS:

```
Running 30s test @ http://10.1.1.1/404-notfound
  6 threads and 6 connections
  Thread Stats   Avg      Stdev     Max   +/- Stdev
    Latency   323.40ms  297.55ms 912.69ms   76.67%
    Req/Sec     4.78      3.29    10.00     44.44%
  90 requests in 30.09s, 10.55KB read
  Socket errors: connect 0, read 1, write 0, timeout 0
  Non-2xx or 3xx responses: 90
Requests/sec:      2.99
Transfer/sec:     358.94B
Running 30s test @ http://10.1.1.1/LICENSE
  6 threads and 6 connections
  Thread Stats   Avg      Stdev     Max   +/- Stdev
    Latency    32.91ms   21.99ms 343.58ms   94.29%
    Req/Sec    21.64      7.33    50.00     60.57%
  2980 requests in 30.10s, 3.46MB read
  Socket errors: connect 0, read 11, write 0, timeout 0
Requests/sec:     99.00
Transfer/sec:    117.86KB
Running 30s test @ http://10.1.1.1/merged-flash.bin
  1 threads and 1 connections
  Thread Stats   Avg      Stdev     Max   +/- Stdev
    Latency     0.00us    0.00us   0.00us     nan%
    Req/Sec     0.00      0.00     0.00    100.00%
  11 requests in 30.08s, 11.92MB read
  Socket errors: connect 0, read 0, write 0, timeout 11
Requests/sec:      0.37
Transfer/sec:    405.98KB
```

stooged, 2022-03-09:
added littlefs support as requested.
you can enable it by setting `#define USELFS true` and selecting a SPIFFS partition.

closed 2022-03-09 by stooged

OpenGG, 2022-10-11:
> What you mention applies to an esp32 s2 mini? I read this comment and
> proceeded to modify my sketch and enabled the littlefs esystem... since my
> s2 mini was configured by default and sometimes when loading the JB I got
> an insufficient memory message or it failed when executing the exploit.
> With this modification I have done 4 tests (two restarting the console and
> two turning it off and on) and the execution of the exploit works quite
> well and the execution times are stable between 18 to 24 seconds. On the
> other hand, with spiffs, the times were irregular and it tended to fail
> often.

Probably so.
Spiffs is slow and can lead to unsuccessful load of the exploit page or
resources.
The performance boost of littlefs can help.