stooged/ESP32-Server-900u #49: powersave mode?

OpenGG, 2022-03-11:
Playing for hours, the chip pick up some heat.
Is it worthy to implement some kind of sleep mode?

stooged, 2022-03-12:
added the following option

```

                       // enable esp sleep [ true / false ]
#define ESPSLEEP false // this will put the esp board to sleep after [TIME2SLEEP] minutes
                       // to wake the board up you will need to reboot the console or unplug/replug the esp board or press the reset button on the board.
                       
#if ESPSLEEP
#define TIME2SLEEP 30 // minutes, the esp will goto sleep after this amount of time passes since boot.
#endif

```

closed 2022-03-12 by stooged

neo305, 2022-03-12:
Awesome thanks!!!! Much needed :)