stooged/ESP32-Server-900u #50: small request

EchoStretch, 2022-03-15:
can you make a light version with no website or goldhen fan and default stuff like spiff partition with webserver and exfathax. need it as small as you can get it.lol

stooged, 2022-03-15:
this has everything stripped out apart from the usb functions, exfathax, the redirect for the console user guide and the connect test response for windows.

you will need to add everything in the spiffs storage which will include the exploit loader, goldhen and any html needed to display payload lists etc.

https://www.dropbox.com/s/qj4xwsp3vs6q8ri/ESP32_Server_900u.zip?dl=1

closed 2022-03-15 by stooged

EchoStretch, 2022-03-15:
thanks so much stooged

EchoStretch, 2022-03-15:
Works perfect thanks again

OpenGG, 2022-03-15:
@EchoStretch 
[Modular ESP32-Server-900u](https://github.com/OpenGG/ESP32-Server-900u/tree/main/ESP32_Server_900u), for more flexible hacking if you like.
Almost all assets including usb.bin moved to littlefs, can be replaced without compiling or flashing.

@stooged 
I worked weeks to configure the CI.
Now [github actions](https://github.com/OpenGG/ESP32-Server-900u/blob/main/.github/workflows/main.yml) can compile ESP32 firmware on the fly.
Think you might be interested?

geovane833, 2022-03-30:
voce tem ai algum projeto com goldhen 2.2.2 e ps4 trainer juntos? para esp32 s2 mini