stooged/ESP32-Server-900u #51: Issue with SDCard & files?

DevEase, 2022-03-23:
Im having a weird issue when i upload files to the board it will do all uploads fine but crashes at the end. once rebooted  i can go to admin menu fine and it displays the correct filesystem size and storage used but when i click on file manager or main it will crash,

stooged, 2022-03-23:
are you uploading a large amount of files(200+) ?

DevEase, 2022-03-23:
no i tried doing 50 at first but then formatted again and tried just 12 and same issue but it works when i do 7 files weird

stooged, 2022-03-23:
i will have to do some tests.
what size are the files approximately or the size of the largest file?

DevEase, 2022-03-23:
uh well i tried different files and as soon as i go past 7 it fails heres a screencap of the uart output from my external uart  
[Uart Output Screenshot](https://i.imgur.com/Cg2KqJf.png)  Also when i click main on admin or go directly to the url with the 7 files it does not load and crashes but filemanager shows the 7 files more files then that does not show and crashes [File Manager Screenshot](https://i.imgur.com/JVEyPfk.png) & the files on the storage via file explorer [Card Contents Screenshot](https://i.imgur.com/59971as.png)

stooged, 2022-03-23:
what filesystem is your sd card formatted with, fat32?

DevEase, 2022-03-23:
yes fat32

DevEase, 2022-03-24:
okay, ill give it a shot right now.  im also going to just download everything once again and make sure. 
ill post once i have a update for you.

DevEase, 2022-03-24:
Okay, so the payloads from psxhax work fine but some of mine have issues.  im not sure what i did honestly but it works now. i appreciate your help :)

closed 2022-03-24 by DevEase