stooged/ESP32-Server-900u #56: Admin Page

neo305, 2022-05-04:
Hi any idea why my admin page went from this to this? 
[image]
[image]

stooged, 2022-05-05:
style.css is not loading

https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L965

https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/Pages.h#L224

neo305, 2022-05-05:
> style.css is not loading
> 
> https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L965
> 
> https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/Pages.h#L224
> 
> try navigate to http://10.1.1.1/style.css and see if it shows data.

when i do this page shows
[image]

neo305, 2022-05-05:
[image]

stooged, 2022-05-05:
that style.css contains code that is not from this repo.
it probably has errors which is preventing the css loading correctly in the browser.

closed 2022-05-05 by stooged

neo305, 2022-05-05:
> that style.css contains code that is not from this repo.
> it probably has errors which is preventing the css loading correctly in the browser.

Ok but that’s the same file from first pic so it works. Only difference is the first pic is on a FeatherS2 and the 2nd pic that’s not working is on a S2 Mini

neo305, 2022-05-05:
Or sorry same S2 Mini board used on both pics. The only difference I think is in the 1st working I used LittleFS and 2nd pic not working I have to use FatFS

stooged, 2022-05-05:
the second picture has a custom style.css file.
the hardcoded pages in the repo work regardless of the filesystem used because the internal pages load from memory not the storage.

if you create a custom css file and it has coding errors it can fail to load which will mess up a page that requires it.

the file is being served by the esp board because you can access it using http://10.1.1.1/style.css so that should rule out the esp board and sketch.

i would delete the custom css file and see if the page loads with the internal css

this is the css code for the internal css file
https://pastebin.com/H9x8Ew6G

neo305, 2022-05-05:
> the second picture has a custom style.css file.
>
> the hardcoded pages in the repo work regardless of the filesystem used because the internal pages load from memory not the storage.
>
>
>
> if you create [...]

Thanks so much. Would you be willing to take a look at my Style.css file cuz I'm not sure what and where I need to edit. I can't delete it cuz then my host page would look like shit lol

stooged, 2022-05-05:
upload it to pastebin and post the link here

stooged, 2022-05-05:
Error, this is a private paste or is pending moderation. If this paste belongs to you, please login to Pastebin to view it.

neo305, 2022-05-05:
> Error, this is a private paste or is pending moderation. If this paste belongs to you, please login to Pastebin to view it.

like I said my first time on it but I made it Public https://pastebin.com/EJqgLHLu via @pastebin

neo305, 2022-05-05:
> Error, this is a private paste or is pending moderation. If this paste belongs to you, please login to Pastebin to view it.

maybe cuz I put it in a folder? or cuz I just signed up? made another one https://pastebin.com/BQRLuKzG via @pastebin try that please

stooged, 2022-05-05:
Still getting the same error.
If you created an accout it might be pending approval.

I always just paste as a guest(no login)

neo305, 2022-05-05:
> Still getting the same error. If you created an accout it might be pending approval.
> 
> I always just paste as a guest(no login)

i try that and get this error as soon as I select create paste. wow first time on pastebin and I hate it already lol
This page is no longer available. It has either expired, been removed by its creator, or removed by one of the Pastebin staff.

neo305, 2022-05-05:
https://drive.google.com/file/d/138UR0pwMulzTAZUFPCao1eGDMETge5My/view?usp=sharing

neo305, 2022-05-05:
> Still getting the same error. If you created an accout it might be pending approval.
> 
> I always just paste as a guest(no login)

will this work?
https://drive.google.com/file/d/1hIMYFcX44dhK7y6PUDWKonT4yK6QBQFE/view?usp=sharing

stooged, 2022-05-05:
That worked.

Your css file is missing all of the required css for the admin page add the code from the pastebin link i posted to your css and try it.

neo305, 2022-05-05:
> That worked.
> 
> Your css file is missing all of the required css for the admin page. add the code from the pastebin link i posted to your css and try it.

I noticed that too so that’s what I did but then it changes my host page too much so something is conflicting

neo305, 2022-05-05:
> That worked.
> 
> Your css file is missing all of the required css for the admin page. add the code from the pastebin link i posted to your css and try it.

that or I am just not adding it correctly, can you please try to add it for me and send

neo305, 2022-05-05:
I have no idea how this happened cuz I have never edited that file before and it worked before as you see from pic

stooged, 2022-05-05:
I dont have access to my pc atm but i tried to add them on my phone, try this code in the css file

```
html, body{
margin:auto;
font-family:sans-serif;
font-size:20px;
font-weight:lighter;
text-align:left;
color:#FFFFFF;
background-size:cover;
}
.modal {
  display: none; /* Hidden by default */
  position: fixed; /* Stay in place */
  z-index: 1; /* Sit on top */
  padding-top: 100px; /* Location of the box */
  left: 0;
  top: 0;
  width: 100%; /* Full width */
  height: 100%; /* Full height */
  overflow: auto; /* Enable scroll if needed */
  background-color: rgb(0,0,0); /* Fallback color */
  background-color: rgba(0,0,0,0.4); /* Black w/ opacity */
}

/* Modal Content */
.modal-content {
  font-family: BFARNAZ;
  font-size: 18px;
  position: relative;
  background-color: #dddddd;
  margin: auto;
  padding: 0;
  border: 1px solid #888;
  width: 50%;
  box-shadow: 0 4px 8px 0 rgba(0,0,0,0.2),0 6px 20px 0 rgba(0,0,0,0.19);
  -webkit-animation-name: animatetop;
  -webkit-animation-duration: 0.4s;
  animation-name: animatetop;
  animation-duration: 0.4s
}
.vid
{
    object-fit: inherit;
    position: fixed;
    right: 0;
    z-index: -1;
    bottom: 0;
    min-width: 100%;
    min-height: 100%;
    opacity: 0.4;
}
.bg-darker {
    background-color: #24272b !important;
}
		.button {
		  background:rgba(0,0,0,0.5);
		  box-shadow: 0 5px 20px 0 white;
		  border: 2px solid black;
		  border-radius: 10px;
		  color: white;
		  text-decoration: none;
		  width: 200px;
          height: 40px;
		  display:inline-table;
		  font-family: system-ui;
		  text-align: center;
		}

[... 605 lines cut ...]
-webkit-animation: spin 2s linear infinite;  
animation: spin 2s linear infinite;
}
  
@-webkit-keyframes spin {
0%{ -webkit-transform: rotate(0deg); }
100% { -webkit-transform: rotate(360deg); }
}

@keyframes spin {
0% { transform: rotate(0deg); }
100% {transform: rotate(360deg); }
}

#status {
font-size: 16px;
font-weight: normal;
}

#msgfmt {
font-size: 16px;  
font-weight: normal;
}

```

neo305, 2022-05-05:
ok when I do the only problem I can see for now is it pushes all the icons up almost half a page. it goes from this
[image]
to this when I add what you said from above
[image]

neo305, 2022-05-05:
Maybe I am just pasting it in the wrong area?

neo305, 2022-05-05:
Ok thanks I added all but body part. Flashing now. Cuz if I add body then I have 2 body and that’s why my page gets pushed up but if I leave it out then page looks the same how it should.

neo305, 2022-05-05:
almost got it lol
[image]

neo305, 2022-05-05:
this is what it looks like when I only use what you said above. anyway to get my settings button back to green?
[image]

stooged, 2022-05-05:
The simplest way is to rename your original custom css file to style2.css and change the html payload menu page to use style2.css

neo305, 2022-05-05:
> The simplest way is to rename your original custom css file to style2.css and change the html payload menu page to use style2.css make sure there is no "style.css" file on the storage and the internal css will be used for the admin menus.

Yes!!!!!!!!! Thank You!!!!!!!!!!1 that was the easiest way lol while I have you can you tell me how fix all my errors please lol
https://drive.google.com/file/d/1lpxOXfgqbSpGERGQwk46k0OGklP6vL6L/view?usp=sharing
[image]

neo305, 2022-05-05:
> The simplest way is to rename your original custom css file to style2.css and change the html payload menu page to use style2.css
> make sure there is no "style.css" file on the storage and the internal css will be used for the admin menus.

Ok spoke too soon. Do I have to edit anything else? Cuz it loaded your page and not mine in user guide.

stooged, 2022-05-05:
you should not need to change anything but the css file name, you are leaving the original style.css to be used for the esp boards admin files and you use style2.css for your main payload menu.

the user guide will load index.html and if you have no index.html on the storage it will load the internal index.html page.

those errors/warnings you are getting are coming from a library in "C:\Users\laz30\Documents\Arduino\libraries\WiFi\src\utility\"

it looks like it has issues so update it or remove it or just leave it if the esp board is working fine with wifi.

neo305, 2022-05-05:
> you should not need to change anything but the css file name, you are leaving the original style.css to be used for the esp boards admin files and you use style2.css for your main payload menu.
> 
> the user guide will load index.html and if you have no index.html on the storage it will load the internal index.html page.
> 
> those errors/warnings you are getting are coming from a library in "C:\Users\laz30\Documents\Arduino\libraries\WiFi"
> 
> it looks like it has issues so update it or remove it or just leave it if the esp board is working fine with wifi.

I know that’s all I did was change my file to Style2.css and now it won’t load my index page it loads yours.

stooged, 2022-05-06:
If it is loading the internal index page then there is no index.html on the storage.
It only loads the internal page when the index.html file is not found on the storage.

Check the filesystem and partition info and reflash your html to the board.

stooged, 2022-05-06:
set the filesystem to spiffs, it is the best option to use on low memory devices because it has less overhead for each file.

fatfs and littlefs were added because people asked for them but for the usage case of providing a simple webserver for the ps4 they offer no real benefit and consume more memory per file.

i only use spiffs on my board for my ps4.

the only real reason you would need to use the other file systems is if you are using directories within your hosts html files.

the issue you are having now is storage related, the esp board cannot find **index.html** in the storage so it is loading the data from **progmem**.

you will see in the code
```server.onNotFound([](AsyncWebServerRequest *request){```

the server not found function is triggered when the file you are attempting to load is not in the storage.

it then checks the path and finds you are looking for index.html 
```
    if (path.endsWith("index.html") || path.endsWith("index.htm") || path.endsWith("/"))
    {
        AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", index_gz, sizeof(index_gz));
        response->addHeader("Content-Encoding", "gzip");
        request->send(response);
        return;
    }
```

and it sends the hard coded **index_gz** from the **progmem**

https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/Pages.h#L12

neo305, 2022-05-06:
Awesome thanks I’ll give it a try

neo305, 2022-05-06:
> set the filesystem to spiffs, it is the best option to use on low memory devices because it has less overhead for each file.
>
> fatfs and littlefs were added because people asked for them but for the [...]

here is what I get wh
[image]
en I try Spiffs

neo305, 2022-05-06:
> set the filesystem to spiffs, it is the best option to use on low memory devices because it has less overhead for each file.
>
> fatfs and littlefs were added because people asked for them but for the [...]

Do you have a S2 Mini? wanna see if you can get it all too work?
https://drive.google.com/file/d/1OaEJLNWynRcS5HlVJM8_sM4nOtDac_DK/view?usp=sharing

neo305, 2022-05-06:
Ok this is blowing my mind. So I got Spiffs to flash, but when I load admin page which looks good, I load file manager it says no files. And info show like 3mb free space. So wtf did it flash? So I upload all files. And everything works perfectly. Now my only problem is that I can’t create a bin file 😝

stooged, 2022-05-06:
in the code you uploaded it has `#define USEFAT true`

if you set the partition in the board settings to spiffs and did not change that to false it will have issues

neo305, 2022-05-06:
> in the code you uploaded it has `#define USEFAT true`
> 
> if you set the partition in the board settings to spiffs and did not change that to false it will have issues

Right that was an old upload but I got spiffs to flash and yeah changed true and false but when I go to smashing page it says no files. So I upload them and everything works fine. So now my only problem is it won’t create a bin file after flash. Cuz there’s no files lol

stooged, 2022-05-06:
index loaded for me and the admin page loaded

i did have to use the no ota 1mb app / 3mb spiffs partition because your files are 1.8mb in size
you are including goldhen in your files so you should set int hen to false to free up used space
`#define INTHEN false`

[image]
[image]

stooged, 2022-05-06:
it also works fine using fatfs.

neo305, 2022-05-06:
> it also works fine using fatfs.

so you're saying this should work?
[image]

neo305, 2022-05-06:
> index loaded for me and the admin page loaded
> 
> i did have to use the no ota 1mb app / 3mb spiffs partition because your files are 1.8mb in size you are including goldhen in your files so you should set int hen to false to free up used space `#define INTHEN false`
> 
> [image]
> 
> [image]

Awesome thanks so much!!!!!! so....
[image]