stooged/ESP32-Server-900u #57: Bottom file incorrect

medo145, 2022-05-12:
The bottom stl file seems to be incorrect.

stooged, 2022-05-12:
all the cases were printed on my printer and tested with the boards i have.

https://github.com/stooged/ESP32-Server-900u/tree/main/3D_Printed_Cases/UM_FeatherS2
[image]

boards could vary in size which will require editing the .stl files and the easiest way for most people is to use [Tinkercad](https://www.tinkercad.com/) and import the .stl file to make changes.

closed 2022-05-12 by stooged

medo145, 2022-05-12:
Take a look at the bottom file, it doesn't look like that at all. It looks like a variation of the top file.

stooged, 2022-05-12:
> Take a look at the bottom file, it doesn't look like that at all. It looks like a variation of the top file.

ahh yes i see what you mean now, i have uploaded the correct file.
thanks for the heads up.

https://github.com/stooged/ESP32-Server-900u/blob/main/3D_Printed_Cases/UM_FeatherS2/UM_FeatherS2_bottom.stl