stooged/ESP32-Server-900u #61: Does not work with ESP32 Arduino 2.0.3

tom2199, 2022-06-12:
It compiles without errors but the access point never shows up. Compiling with 2.0.2 does work as intended. Took me half a day to figure that one out 😂

ahrushetskyi, 2022-07-09:
Does not work with 2.0.4 either
Works only with 2.0.2

stooged, 2022-07-11:
updated my esp32 package to 2.0.4(current) and it compiled and booted fine.
AP works and exfathax via usb emu works fine.

i cant replicate the issue you are having based on the available information.

closed 2022-07-11 by stooged

tom2199, 2022-07-11:
As I'm not the only one having the issue maybe there is something to it? My guess is there are cached files of some sort on your machine and that's why you can't replicate it. I tried on two machines one of them didn't had Arduino IDE installed beforehand (so it was a fresh install) and both of them showed the same issue.

stooged, 2022-07-11:
just used a fresh win10 vm, installed arduino ide and the required libraries including version 2.0.4 of the esp32 board lib, pulled this repo and compiled it and uploaded to a blank board(new) and everything runs as expected.

the ap loads and exploit fires usb(exfathax) when expected.

this was done using all project defaults on a `ESP32S2 Dev Module` compatible board