stooged/ESP32-Server-900u #71: Failed to trigger exploit

Rippa84, 2022-09-19:
Hi I have a problem triggering the exploit.
I uploaded the original code/scetch with no changes (only changing the partion scheme to "Default 4MB with spiffs") to my Adafruit QT PY ESP32-S2. After loading the Main Page and trying to load Goldhen I will allways get this error message.
[image]

The Main Page first says something about loading the exploit and then performing exfathax. Then the error appears.
Tried restarting and power off and different USB Ports and Cables, but no changes.

stooged, 2022-09-19:
exfathax is probably not being triggered on the console.

look for the usb error message during the exploit process
[image]

Rippa84, 2022-09-20:
I had to set the USB CDC on Boot from Enabled to Disabled and upload the scetch again. Working like a charm now. Thanks for your work.

closed 2022-09-20 by Rippa84