stooged/ESP32-Server-900u #73: Does not connect to home Wi-Fi network

eemcm2, 2022-10-03:
I have tried with two wemos s2 mini boards connecting them to 2 different home Wi-Fi networks, from the administration page, the SSID and password of the home Wi-Fi are configured, but the PS4 does not get internet.

From the configuration of my router I can see that the card is connected to the network, but it does not give internet to the console.

stooged, 2022-10-03:
the esp32 board does not bridge the connections so if you want the ps4 to have internet you need to connect the ps4 to your home wifi and connect the esp32 board to your home wifi and then connect to the esp32 board from the ps4 through your home wifi to perform the exploit.

closed 2022-10-03 by stooged

eemcm2, 2022-10-03:
I did what you mention (I think I understood it) I connected the board to my home Wi-Fi network, then I connected the console to my home Wi-Fi network.  I entered the browser to the address 10.1.1.1 but it does not work.  it says there is no connection.

stooged, 2022-10-03:
Try http://ps4.local

eemcm2, 2022-10-03:
I'll keep testing as going into ps4.local didn't work either... the sketch code is almost identical to Github's, however the only thing I added was the option to enable the LED light on the board.  I'll write the sketch on the board again but without that code... I'll comment how it went.

eemcm2, 2022-10-04:
I finally managed to get an internet connection on my PS4 and get the esp32 server working.  The only way to make it work was the following:

From the board's admin page, I've set up my home wifi network.  Once connected, from some Wi-Fi network monitoring app, the connected board appeared and showed me an IP.  I entered that IP in my phone's browser and connected to the server!  Finally, I entered the IP into the ps4 browser and I was finally able to run the exploit and best of all, I do it with the console connected to the 5G Wi-Fi network at home!