stooged/ESP32-Server-900u #74: PS5 userguide

EchoStretch, 2022-10-09:
Im trying to setup user guide on ps5 to redirect to it as we don't have a browser .. 
ps5 user guide goes to ecode.np.cs.playstation.net, can it be redirected to the 10.1.1.1?

EchoStretch, 2022-10-09:
trying to have ps4 9.00 with the new ps5 added too

stooged, 2022-10-09:
yes, the dns server redirects all domains to the set server ip but with the ps5 it is a https request so you need a https webserver and you then redirect the https request to a http request which takes the browser out of https mode

https://github.com/stooged/PS5-Server32/blob/main/PS5_Server32/PS5_Server32.ino#L867

you can just run everything under https but i find it better to just get out of https

closed 2022-10-09 by stooged

EchoStretch, 2022-10-09:
yea i was looking at ps5 server32.. its fine for ps5 side of things but when it comes to auto usb emulation it doesnt have it for 9.00.. so it would have to be set to insert usb each time.

EchoStretch, 2022-10-09:
Was looking to have both on the s2 and feathers2 so no one needs 2 boards for ps4 and ps5.. its fine for the ps5 server 8266.

EchoStretch, 2022-10-09:
is that possible or just a pain in *** .. i was looking over the code but it looks like a headache. haha

stooged, 2022-10-09:
here you go, i added the https redirect 

https://github.com/stooged/ESP32-Server-900u/tree/main/ESP32_Server_900u_HTTPS

EchoStretch, 2022-10-09:
sweet thx

stooged, 2022-10-09:
the redirect is on this line if you want to point it somewhere other than index.html

https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u_HTTPS/ESP32_Server_900u_HTTPS.ino#L632

EchoStretch, 2022-10-09:
ultiple libraries were found for "WiFi.h"
ESP32_Server_900u_HTTPS:8:10: fatal error: HTTPSServer.hpp: No such file or directory
 Used: C:\Users\EchoStretch\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi
 #include <HTTPSServer.hpp>
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
          ^~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
HTTPSServer.hpp: No such file or directory

stooged, 2022-10-09:
yeah you need to install the https server lib

https://github.com/stooged/esp32_https_server

stooged, 2022-10-09:
take the one from my repo as it has had the sha patch, the original repo is out of date.

EchoStretch, 2022-10-09:
In file included from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPSServer.hpp:11,
                 from C:\Users\EchoStretch\Desktop\7in1\ESP32S2\ESP32_Server_900u_HTTPS\ESP32_Server_900u_HTTPS.ino:8:
C:\Users\EchoStretch\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32s2/include/openssl/include/openssl/ssl.h:8:2: warning: #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead" [-Wcpp]
 #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead"
  ^~~~~~~
In file included from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/ConnectionContext.hpp:8,
                 from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPRequest.hpp:10,
                 from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPSCallbackFunction.hpp:4,
                 from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/ResourceNode.hpp:7,
                 from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPServer.hpp:20,
                 from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPSServer.hpp:15,
                 from C:\Users\EchoStretch\Desktop\7in1\ESP32S2\ESP32_Server_900u_HTTPS\ESP32_Server_900u_HTTPS.ino:8:
C:\Users\EchoStretch\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32s2/include/openssl/include/openssl/ssl.h:8:2: warning: #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead" [-Wcpp]
 #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead"
  ^~~~~~~
In file included from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPResponse.hpp:12,
                 from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPSCallbackFunction.hpp:5,
                 from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/ResourceNode.hpp:7,
                 from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPServer.hpp:20,
                 from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPSServer.hpp:15,
                 from C:\Users\EchoStretch\Desktop\7in1\ESP32S2\ESP32_Server_900u_HTTPS\ESP32_Server_900u_HTTPS.ino:8:
C:\Users\EchoStretch\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32s2/include/openssl/include/openssl/ssl.h:8:2: warning: #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead" [-Wcpp]
 #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead"
  ^~~~~~~
In file included from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPSConnection.hpp:9,
                 from C:\Users\EchoStretch\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPSServer.hpp:22,
                 from C:\Users\EchoStretch\Desktop\7in1\ESP32S2\ESP32_Server_900u_HTTPS\ESP32_Server_900u_HTTPS.ino:8:
C:\Users\EchoStretch\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32s2/include/openssl/include/openssl/ssl.h:8:2: warning: #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead" [-Wcpp]
 #warning "OpenSSL component will be removed from ESP-IDF in v5.0, please use esp_tls instead"
  ^~~~~~~

stooged, 2022-10-09:
dont worry about the warnings, like i said the lib is out of date but works enough to perform the handshake and redirect to http

EchoStretch, 2022-10-10:
ok after doing a few tests..
takes up more space so harder to get everything onto esp32s2 so i will have remove some payloads
but moved over to the feathers2
loads great on ps4 and does jailbreak but when on ps5 takes few seconds to load host and when I hit jailbreak it just stops at the spinning bar and doesn't do the jailbreak

stooged, 2022-10-10:
once you bounce to http everything is handled by the [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer)

you might want to test on a pc that all of the exploit files are loading and that they are loading fully