stooged/ESP32-Server-900u #77: How to ESP32-S2 to FeatherS2

neo305, 2022-11-02:
Hi can you help me. I am trying to port over a ESP32-S2 host over to a FeatherS2 so I can utilize the 16mb. I am guessing it is failing cuz of the partition scheme. on ESP32-S2 it uses a LittleFS and I don't see one like that on FeatherS2. Please help, if you need more info lmk

stooged, 2022-11-02:
the project has been tested and works on the feather s2 boards.

you set the partition scheme as ```16M Flash (2MB APP/12.5MB FATFS)```

and then set the definition to use FatFs in the code ```#define USEFAT true```

closed 2022-11-02 by stooged

neo305, 2022-11-03:
Thank you so much. any idea how I get that scheme?
[image]

stooged, 2022-11-03:
FATFS and FAT are basically the same so just choose ```16M Flash (2MB APP/12.5MB FAT)``` in your list