stooged/ESP32-Server-900u #79: Add browser cache option

eemcm2, 2022-11-14:
Can you add the option to save the content of esp32 and Index in the ps4 browser cache?

stooged, 2022-11-14:
you can do that by creating your own [cache manifest](https://en.wikipedia.org/wiki/Cache_manifest_in_HTML5) file and creating your own index.html and adding the manifest attribute to it.

closed 2022-11-14 by stooged

eemcm2, 2022-11-15:
I'm pretty new at this and I'm practically learning by myself.

I added the cache function to the index and created the file manifest. I loaded them on the esp32 and on the ps4 it saves the cache and loads the exploit, but I can't get it to load goldhen.

stooged, 2022-11-15:
make sure you cache all of the required files
```
index.html
menu.html
loader.html
payloads.html
style.css
goldhen.bin
```

eemcm2, 2022-11-17:
in index.html I add the following: 
<html manifest="index.cache">

in cache.manifest I add the following:

CACHE MANIFEST
index.html
menu.html
loader.html
payloads.html
style.css
goldhen.bin

I run the server, apparently it caches and runs the webkit exploit, but when loading goldhen it doesn't do anything, if I remember, a null load message appears

stooged, 2022-11-18:
you will need to make sure that each file is cached correctly, try unplugging the board so there is no wifi server and see what loads, maybe try and load each file individually in the ps4 browser to see if they are cached.

it wont load the kexploit while it is unplugged but you can still use it to test the files.

eemcm2, 2022-11-19:
I tried again to make save in cache work and it did not work, I disconnected the board and it did not save in cache. I give up, anyway, thanks for your help.