stooged/ESP32-Server-900u #80: Request

neo305, 2022-11-15:
He man sorry to bother you but would you have time to make a how to create bin file using flash download tool? 
https://gbatemp.net/threads/esp8266-how-to-make-a-ps4-exploit-bin-file-to-host-and-flash.576934/page-2
I use to be able to do esp8266 boards but this esp32-s2 is difficult

stooged, 2022-11-15:
i am not quite sure what you mean...

are you looking for a compiled binary file to flash to the esp board?

neo305, 2022-11-15:
No i mean how to create a bin file from sketch. Cuz I can flash my sketch to board np but would like to convert it into a bin file to flash later as a faster method.

stooged, 2022-11-15:
you create a bin file from a sketch by selecting "export compiled binary" in the arduino ide
it will be created in the same folder the sketch is in.

[image]

neo305, 2022-11-15:
ok please tell me it is not that easy lol I try that and get a 847kb file ESP32_Server_900u.ino.esp32s2.bin. What are the steps? I select Upload then in Tools select Sketch Data Upload, then under Sketch select compile. Correct?

neo305, 2022-11-15:
Sketch uses 866950 bytes (82%) of program storage space. Maximum is 1048576 bytes.
Global variables use 46912 bytes (14%) of dynamic memory, leaving 280768 bytes for local variables. Maximum is 327680 bytes.

stooged, 2022-11-15:
ESP32_Server_900u.ino.esp32s2.bin

Is the compiled bin file.

You could use that bin to to an ota update or convert it to a .hex file and use tools to upload it to the board over serial.

There is another way to get the individual bins used for boot, code, partitions and upload them all separately but it is all much more complicated than a single bin

https://www.aranacorp.com/en/generating-and-uploading-bin-files-to-an-esp8266/amp/

neo305, 2022-11-15:
Thanks. Last thing. Any plans for the new RaspberryPi Pico W? Or can esp32 work on it?

stooged, 2022-11-18:
https://github.com/stooged/Pico-Server-900

closed 2022-11-18 by stooged