stooged/ESP32-Server-900u #85: wifi issue with esp32 s2 mini

ifelixculpa, 2022-12-09:
hi i have this card and  keep getting connection to wifi is lost ..please help
[img]https://i.imgur.com/HKVFLe1.png[/img]

eemcm2, 2022-12-09:
La placa genera el punto de acceso wifi? Puedes revisar el tiempo de sueño profundo, ya que su lo tienes activo con una cantidad determinada de tiempo, la placa se apaga. De todas formas, es preferible que se apague después de unos diez minutos, para evitar sobre calentamiento.

ifelixculpa, 2022-12-10:
> La placa genera el punto de acceso wifi? Puedes revisar el tiempo de sueño profundo, ya que si lo tienes activo con una cantidad determinada de tiempo, la placa se apaga. De todas formas, es preferible que se apague después de unos diez minutos, para evitar sobre calentamiento.
> 
> Recuerda que la placa se utiliza solo para cargar el exploit.

ohhh, eso lo explica, me preocupaba que la placa estuviera defectuosa... muchas gracias señor, me siento aliviado ahora, muchas gracias :) utilizo el host leaful que está alojado en 10.1.1.2 ¿cómo debo acceder a la configuración para eso?

ohhh that explains it , i got worried about board being defective...thanks alot sir i am relieved now thanks alooooot :) i use leaful host which is hosted at 10.1.1.2 how shall i aceess the settings for that ?

eemcm2, 2022-12-11:
Si usas PS phive, ayer lanzaron la versión 3 y creo que tiene un acceso directo a la configuración.

Yo uso un mod hecho por mi del host v5 de leeful. En mi caso, guarda en host en el caché del navegador, activa la luz de estado de la placa y activa psram. 

Si quieres, puedes instalarlo desde mi instalador online, los textos del host están en español e incluye la última versión de Goldhen 2.2.5b8.

eemcm2.guthub.io/225b8

ifelixculpa, 2022-12-12:
> Si usas PS phive, ayer lanzaron la versión 3 y creo que tiene un acceso directo a la configuración.
> 
> Yo uso un mod hecho por mi del host v5 de leeful. En mi caso, guarda en host en el caché del navegador, activa la luz de estado de la placa y activa psram.
> 
> Si quieres, puedes instalarlo desde mi instalador online, los textos del host están en español e incluye la última versión de Goldhen 2.2.5b8.
> 
> eemcm2.github.io/auto225b8
> 
> Puedes acceder a la configuración del servidor desde 10.1.1.1/admin.html

impresionante hermano en uso o muchas gracias hermano :)

closed 2023-01-06 by stooged