stooged/ESP32-Server-900u #90: ESP32_Server_900u_HTTPS seems to be too big for Lolin S2 Mini

jacoghi, 2023-01-02:
Hi stooged,
Just tried your new HTTPS sketc, however, it fails for the Lolin S2 Mini
Sketch uses 1507914 bytes (115%) of program storage space. Maximum is 1310720 bytes.
Global variables use 58372 bytes (17%) of dynamic memory, leaving 269308 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

eemcm2, 2023-01-03:
just configure arduino as the attached image, also don't update your boards to 2.0.6 as it crashes a lot.

[image]

jacoghi, 2023-01-03:
Hi @eemcm2, thanks for the comment, do tou think it could be related to 2.0.6 itself? Those are exactly the settings I used and the USB would not work on sending the payload at all, I had to change it to thia deneyap mini (it worked normally with 1.8.x selecting Lolin, so maybe be indeed the IDE due to USB to UART or something like that?) to get it working. Regarding the https server, I tried with configurations for both boards and both complained about the max size of the board, I could get it to work with the standard "U" ino.

jacoghi, 2023-01-03:
Aha, that makes sense, since I'm using AutoHen with the included Golhen, but no fanmod. Guess I'll try it again later and see how it goes. Thanks for the feedback, man!

eemcm2, 2023-01-03:
that must have generated your problem, better use autohen and disable inthen.  Then you can upload the version of goldhen you want from 10.1.1.1/admin.html in file upload, remember to change the file name to "goldhen.bin"

stooged, 2023-01-06:
in your initial post it shows this

Sketch uses **1507914** bytes (115%) of program storage space. Maximum is **1310720** bytes. 

this means the partition scheme you have selected is not large enough to fit the sketch.
you can disable the inbuilt hen binary "INTHEN" or you can change the selected partition scheme to allow a larger APP partition.
you have a partition selected that is **1.2MB APP**  if you select a partition with something like **2MB APP** the sketch will fit.

on a side note you do not need to delete the fan.h and goldhen.h files from the project folder, just set INTHEN and FANMOD to false and the compiler will be instructed to ignore those files and all code attributed to those functions.

closed 2023-01-06 by stooged