stooged/ESP32-Server-900u #91: File Manager Tab

neo305, 2023-01-07:
stooged I can’t figure out why or how I broke the file manager tab in admin page. Can you take a look at my project and see if you can tell me how to fix it please?

stooged, 2023-01-07:
you dont need to open a new issue each time, you can keep making comments in closed issues but if you upload it i can have a look to see if there is any reason the **fileman.html** is not loading

closed 2023-01-07 by stooged

neo305, 2023-01-07:
Thanks man and sorry about that, I wasn’t sure if you still looked at closed cases lol can I email it to ya? I don’t want everyone to have it lol laz305@gmail.com

neo305, 2023-01-07:
or better yet how can I fix this? I'm on 2.0.5 should I switch to 2.0.2?

Uploading using selected port: COM3
C:\Users\laz30\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1/esptool.exe --chip esp32s2 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash {upload.erase_cmd} -z --flash_mode dio --flash_freq 80m --flash_size 16MB 0x1000 C:\Users\laz30\AppData\Local\Temp\arduino_build_35154/ESP32_900.ino.bootloader.bin 0x8000 C:\Users\laz30\AppData\Local\Temp\arduino_build_35154/ESP32_900.ino.partitions.bin 0xe000 C:\Users\laz30\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5/tools/partitions/boot_app0.bin 0x10000 C:\Users\laz30\AppData\Local\Temp\arduino_build_35154/ESP32_900.ino.bin 
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address "{upload.erase_cmd}" must be a number
esptool write_flash: error: argument <address> <filename>: Address "{upload.erase_cmd}" must be a number

neo305, 2023-01-07:
ok bro yeah 2.0.2 worked for me. are you able to assist in compiling bin file?
https://drive.google.com/file/d/18ZRvz7STYNtwNqQoBIJRHMmLhzh_7e1A/view?usp=share_link

stooged, 2023-01-07:
the flash log.txt file you posted shows compilation and flashing success.

for the other error it looks like something is wrong in the esptool commandline 
```esptool write_flash: error: argument : Address "{upload.erase_cmd}" must be a number```

it looks like it is passing this as an arg `{upload.erase_cmd}` which should be a number so something is messed up in 2.0.5

neo305, 2023-01-08:
Ok thanks yeah 2.0.2 solved all those issues. Now my problem is auto load golden will work in browser but not user guide. Any ideas?

stooged, 2023-01-08:
the user guide page should be getting redirected to /index.html which would be the same as using the browser and going to /index.html so i am not sure why it would not load.

maybe use a pc webbrowser and goto `http://IPADDRESS/document/us/ps4/index.html` and see what it is trying to load in the network monitor

neo305, 2023-01-08:
Thanks I tried. It goes no where. I’ve always had it like this “document/en/ps4/index.html it’s always worked before but not this time. I have no idea why.

neo305, 2023-01-08:
And Yeah 192.168.1.1/document/en/ps4/index.html just opens a blank white page

neo305, 2023-01-08:
ip/index.html opens the cache page as it should btw

neo305, 2023-01-08:
Ok nm sorry I’m all good man. Thanks for your time.

stooged, 2023-01-08:
by the looks of it you are using cache manifest?

if so the reason your cache is not working is because you need to cache the path `document/en/ps4/` and associated files.
without the board connected there is no redirect for the user guide so the console is trying to operate on that path and if your cache manifest does not have that path cached then it cant load the file from the cache for the given path like `document/en/ps4/index.html`.

you could make a redirect html file and add it to your manifest so any time you attempt to access the user guide it uses a meta redirect or js redirect to jump to /index.html which would be your cached index.html from the ps4 webbrowser.

stooged, 2023-01-08:
i wouldnt bother with cache myself because the board needs to be plugged in for exfathax so it will never be like the older exploits where you can cache everything and then unplug the esp and leave it unplugged.

neo305, 2023-01-08:
Yeah but with sleep mode I can still load all other bins from wake and with them all cache I can load them and don’t need board plugged in unless I want to load Goldhen. Thanks for the info I’ll give it a shot.

stooged, 2023-01-08:
ahh i see, i dont use sleep mode on the esp, i never really saw the point of it but someone asked for it to be added.

most of the issues you will be having will all be related to the cache, you need to make sure you cache all paths and files.
so if you have index.html cached on the path `http://IPADDY/index.html` it wont work if you try to access index.html via `http://IPADDY/document/en/ps4/index.html` as you need to cache it on the path being accessed.

the same will go for all files related to it like images, js, css, bin files etc.
so if i were to cache the function to bounce the user guide to the root so i dont have to cache all of the files and html again i would modify the cache manifest to point the path for the user guide to point to a html file like `redirect.html` and in that i would just have js code to redirect it to the root `/index.html`

then all you would need to do is cache a single file for the user guide `http://IPADDY/document/en/ps4/redirect.html` and that would bounce you to the root `/` where all the files are already cached.

neo305, 2023-01-08:
Last thing man. Hey any clue about this error in Arduino. I basically have everything the same on pc and laptop but on pc after every flash Arduino stays frozen and I have to open task manager and close Java to shut Arduino down.

stooged, 2023-01-08:
the esptool might be locking up causing the whole lot to freeze

maybe have a look at https://github.com/espressif/esptool and see if anyone has had the same issue

neo305, 2023-01-10:
ok stooged I am almost completely done lol just need to figure out this ip situation. ok so i have 3 boards and i want each with different ip and i want to make them simple like 10.1.1.1 and 10.1.1.2 and 10.1.1.3 but anytime I change these numbers then build bin file fails but it'll still flash 100%. Am I missing a location to change the ip or not matching it somewhere?
[image]
[image]

stooged, 2023-01-10:
those values will be replaced by the values in the config that is saved to the board storage so if you are hard coding them then disable the config
https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L51

```#define USECONFIG false ```

or wipe the boards storage using the esp32 filesystem uploader by selecting ```!Erase Flash!```
https://github.com/stooged/arduino-esp32fs-plugin

neo305, 2023-01-11:
Ok thanks I got it. Just to be sure, I was changing the ip and still had it set on true so that’s why building the bin file would fail?

stooged, 2023-01-11:
You said it flashed so for it to flash it should have compiled first.

If it is failing to compile then you are doing something wrong in the code.

neo305, 2023-01-11:
Ok from the pic of my index. I can make those any ip I want right? 10.1.1.1 and 10.1.2.3?

stooged, 2023-01-11:
As long as the ip of the board matches the ip in your html it should all work