stooged/ESP32-Server-900u #92: Cant connect multiple devices to the wifi AP at the same time

Svahnen, 2023-01-08:
I want to connect my pc to the wifi AP too so I can install over the network, I do not want to connect the ESP to my router but instead just switch my PC wifi over to the ESP AP while installing.

Whenever I try this I get a message that the IP is already in use and I have to set my address manually to IP 10.1.1.3 subnet 255.255.255.0 and gateway/router to 10.1.1.1.

I am guessing there is no handler for giving out multiple IP addresses at once.

(I also read somewhere that the ESP goes to sleep after a while, as its only needed when initializing the hack, but for my use case this should be disabled, anyone know where to turn this off in the code?)

Svahnen, 2023-01-08:
//ESP sleep after x minutes
boolean espSleep = false;
int TIME2SLEEP = 30;  // minutes

Found this part on line 76-78, so i guess the sleep part is figured out, now all i need is to figure out how to get some type of DHCP to work for multiple clients

stooged, 2023-01-08:
The `WiFi.softAP()` function should be assigning the ip address range automatically.

There was an update to the `softAPConfig` in 2.0.4 to allow you to set the start ip in the range.

You could try and change this line and add the start ip

https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L735

Add something like this

```
IPAddress DHCP_Start_IP(10, 1, 1, 2);
WiFi.softAPConfig(Server_IP, Server_IP, Subnet_Mask, DHCP_Start_IP);
```

it might fix the issue.

Francesco146, 2024-02-10:
it's quite an old issue, is this still relevant or should be closed?

—— i also take this opportunity to thank you for updating the repo so quickly with PSFree