stooged/ESP32-Server-900u #93: Help to add code to blink the integrated led in ESP32 S2 Mini.

eemcm2, 2023-01-13:
Hello stooged, I wanted to ask for your help to be able to add the code so that the integrated led blinks every 1 second, I already have the code, but I don't know how to insert it into your sketch. I have tried, the blinking of the led works, but the board works unstable, it becomes slow or micro cuts of the wifi signal appear.

I insert the void setup part and it works fine, but I don't know how to insert the void loop part.

void setup() {
  pinMode(LED, OUTPUT);
}

void loop() {
  digitalWrite(LED, HIGH);
  delay(1000);
  digitalWrite(LED, LOW);
  delay(1000);
}

`void loop() {
  if (espSleep && !isFormating) {
    if (millis() >= (bootTime + (TIME2SLEEP * 60000))) {
      //HWSerial.print("Esp sleep");
      esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
      esp_deep_sleep_start();
      return;
    }
  }
  if (hasEnabled && millis() >= (enTime + 15000)) {
    disableUSB();
  }
#if !USESD
  if (isFormating) {
    //HWSerial.print("Formatting Storage");
    isFormating = false;
    FILESYS.end();
    FILESYS.format();
    FILESYS.begin(true);
    delay(1000);
#if USECONFIG
    writeConfig();
#endif
  }
#endif
  dnsServer.processNextRequest();
}`

stooged, 2023-01-14:
its slowing down because you are putting a delay into the loop

delay(1000); in the loop() will halt code execution for 1 second and if you have it 2 times it will halt for 2 seconds so a quick way to get it to flash without using a delay is to use the [millis()](https://www.arduino.cc/reference/en/language/functions/time/millis/) function.

Description
```Returns the number of milliseconds passed since the Arduino board began running the current program```

so you basically make a timer within the loop storing the last time you switched the led state and then adding 1000 milliseconds to that before the next state change and add a boolean to hold the current state of the led.

doing it this way is not perfect and other code execution or delays will interrupt the flashing pattern, if you had a dual core board you could offload the blinking to the second core and it would blink uninterrupted but the esp32-s2 soc is a single core so this is just a quick and dirty way to get it blinking without halting the code execution of other functions.

first setup your globals.
```
int LED = LED_BUILTIN;
boolean ledOn = false;
long ledTime = 0;
```

then in the setup() function set the pin mode to output.
```
pinMode(LED, OUTPUT);
```
 
then in the loop() add the blinky code.
```
  if (millis() >= (ledTime + 1000)) {
   if (ledOn){
    ledOn = false;
    digitalWrite(LED, LOW);
   }
   else
   {
    ledOn = true;
    digitalWrite(LED, HIGH);
   }
    ledTime = millis();
  }

```

stooged, 2023-01-14:
if you are targeting a specific pin for the led and not the builtin led then change this 
`int LED = LED_BUILTIN;` to the pin number like `int LED = 13;`

closed 2023-01-14 by stooged

eemcm2, 2023-01-14:
Let's see if I understood, would I have to add the code in this way to the void loop?

`void loop() {
if (millis() >= (ledTime + 1000)) {
   if (ledOn){
    ledOn = false;
    digitalWrite(LED, LOW);
   }
   else
   {
    ledOn = true;
    digitalWrite(LED, HIGH);
   }
    ledTime = millis();
  }
  if (espSleep && !isFormating) {
    if (millis() >= (bootTime + (TIME2SLEEP * 60000))) {
      //HWSerial.print("Esp sleep");
      esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
      esp_deep_sleep_start();
      return;
    }
  }
  if (hasEnabled && millis() >= (enTime + 15000)) {
    disableUSB();
  }
#if !USESD
  if (isFormating) {
    //HWSerial.print("Formatting Storage");
    isFormating = false;
    FILESYS.end();
    FILESYS.format();
    FILESYS.begin(true);
    delay(1000);
#if USECONFIG
    writeConfig();
#endif
  }
#endif
  dnsServer.processNextRequest();
}`

eemcm2, 2023-01-14:
It works perfect! Also, add code so that when turning on the board it blinks 3 times faster, it ended up like this:

`void setup() {
  //HWSerial.begin(115200);
  //HWSerial.println("Version: " + firmwareVer);
  //USBSerial.begin();

  pinMode(LED, OUTPUT);
  for (int i = 0; i < 3; i++) { //repite el encendido del LED 3 veces
    digitalWrite(LED, HIGH); //enciende el LED
    delay(200); //espera medio segundo
    digitalWrite(LED, LOW); //apaga el LED
    delay(200); //espera medio segundo
  }`

`void loop() {
  if (millis() >= (ledTime + 1000)) {
   if (ledOn){
    ledOn = false;
    digitalWrite(LED, LOW);
   }
   else
   {
    ledOn = true;
    digitalWrite(LED, HIGH);
   }
    ledTime = millis();
  }
  if (espSleep && !isFormating) {
    if (millis() >= (bootTime + (TIME2SLEEP * 60000))) {
      //HWSerial.print("Esp sleep");
      esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
      esp_deep_sleep_start();
      return;
    }
  }
  if (hasEnabled && millis() >= (enTime + 15000)) {
    disableUSB();
  }
#if !USESD
  if (isFormating) {
    //HWSerial.print("Formatting Storage");
    isFormating = false;
    FILESYS.end();
    FILESYS.format();
    FILESYS.begin(true);
    delay(1000);
#if USECONFIG
    writeConfig();
#endif
  }
#endif
  dnsServer.processNextRequest();
}`

eemcm2, 2023-01-14:
It works perfect! Also, add code so that when turning on the board it blinks 3 times faster, it ended up like this:
can you confirm if the void loop part is ok? (sorry for the noob, I'm learning arduino) and the code to make the led blink faster 3 times when turned on? since I'm using gpt chat to support me :D

`void setup() {
  //HWSerial.begin(115200);
  //HWSerial.println("Version: " + firmwareVer);
  //USBSerial.begin();

  pinMode(LED, OUTPUT);
  for (int i = 0; i < 3; i++) { //repite el encendido del LED 3 veces
    digitalWrite(LED, HIGH); //enciende el LED
    delay(200); //espera medio segundo
    digitalWrite(LED, LOW); //apaga el LED
    delay(200); //espera medio segundo
  }`
`void loop() {
  if (millis() >= (ledTime + 1000)) {
   if (ledOn){
    ledOn = false;
    digitalWrite(LED, LOW);
   }
   else
   {
    ledOn = true;
    digitalWrite(LED, HIGH);
   }
    ledTime = millis();
  }
  if (espSleep && !isFormating) {
    if (millis() >= (bootTime + (TIME2SLEEP * 60000))) {
      //HWSerial.print("Esp sleep");
      esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
      esp_deep_sleep_start();
      return;
    }
  }
  if (hasEnabled && millis() >= (enTime + 15000)) {
    disableUSB();
  }
#if !USESD
  if (isFormating) {
    //HWSerial.print("Formatting Storage");
    isFormating = false;
    FILESYS.end();
    FILESYS.format();
    FILESYS.begin(true);
    delay(1000);
#if USECONFIG
    writeConfig();
#endif
  }
#endif
  dnsServer.processNextRequest();
}`

eemcm2, 2023-01-14:
It works perfect! Also, add code so that when turning on the board it blinks 3 times faster, it ended up like this:
can you confirm if the void loop part is ok? (sorry for the noob, I'm learning arduino) and the code to make the led blink faster 3 times when turned on? since I'm using gpt chat to support me :D

```
void setup() {
  //HWSerial.begin(115200);
  //HWSerial.println("Version: " + firmwareVer);
  //USBSerial.begin();

  pinMode(LED, OUTPUT);
  for (int i = 0; i < 3; i++) { //repite el encendido del LED 3 veces
    digitalWrite(LED, HIGH); //enciende el LED
    delay(200); //espera medio segundo
    digitalWrite(LED, LOW); //apaga el LED
    delay(200); //espera medio segundo
  }
```

```
void loop() {
  if (millis() >= (ledTime + 2000)) {
   if (ledOn){
    ledOn = false;
    digitalWrite(LED, LOW);
   }
   else
   {
    ledOn = true;
    digitalWrite(LED, HIGH);
   }
    ledTime = millis();
  }
  if (espSleep && !isFormating) {
    if (millis() >= (bootTime + (TIME2SLEEP * 60000))) {
      //HWSerial.print("Esp sleep");
      esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
      esp_deep_sleep_start();
      return;
    }
  }
  if (hasEnabled && millis() >= (enTime + 15000)) {
    disableUSB();
  }
#if !USESD
  if (isFormating) {
    //HWSerial.print("Formatting Storage");
    isFormating = false;
    FILESYS.end();
    FILESYS.format();
    FILESYS.begin(true);
    delay(1000);
#if USECONFIG
    writeConfig();
#endif
  }
#endif
  dnsServer.processNextRequest();
}
```