stooged/ESP32-Server-900u #94: USB is always on in wiring mode

PAYAMiN, 2023-01-23:
I am using NanoESP32-S2 and had to use wiring mode because I couldn't find any solution to use existed USB-C ports on my board to connect EXFATHAX usb flash. 
Now wireless AP, settings and everything is working fine but still problem with USB is persist. 
In wiring mode (I used ESP32-S2-Saola-1 Wiring Diagram, as seems very similar to my board) USB is recognized but it is always on and make my console get freezed in startup, and exploit is not working because there is no auto connect USB simulation when EXFATHAX is going to be triggered.
Do you have any idea where the issue is?

P.S: I changed USBCONTROL to True in line 14, but I don't have any idea about which usbPin should be used in next line, so I used 1, 2, 3, 4 as well as 19, 20 (the pins D+ D- connect to), but the result was same

stooged, 2023-01-23:
USBCONTROL is only used for esp32 boards, your board is a esp32-s2 so the USBCONTROL flag is ignored.

your board looks like it has 2 usb ports, one is labeled **CH340** and the other is labeled **ESP32**
the port you use to connect to the ps4 is the **ESP32** port, the other port will not work.

when you plug a usb cable into the **ESP32** port it is the same as wiring a cable to pins 19 and 20 so you do not need to wire a usb plug on, just use the **ESP32** port with a cable that supports data transfer(not a charge cable)

if you still have trouble with the exfathax then try and change the upload mode to **internal usb** and flash the board through the ESP32 port not the CH340 port.

[image]

PAYAMiN, 2023-01-24:
Thank you very much @stooged , I followed your instruction and it works like a charm! 
Appreciated boss :)

closed 2023-01-24 by PAYAMiN