stooged/ESP32-Server-900u #97: Pages.h

oresterosso75, 2023-02-26:
I was wondering if it was possible to make changes to this template without necessarily having to recreate an index.html
probably this file needs to be converted to html in order to make changes..
solutions?
Thank's for your job

rynho, 2024-02-04:
The .h file contains the comprised internal pages in integar arrays.
You can extract the integer arrays (i.e., comma sparated integars in curved brackets), and convert them into binary form (e.g., '10' -> '0x0A') using an hex editor, or you can write/find a program to do so.  After that, save each of those binary file into .gz format and extract the page from them.  After your edit, you need to revert the process and update the .h file, this can be facilitated by the bin2h tool included in this project.

Francesco146, 2024-02-10:
i think this integration between html files and header files should be facilitated directly with a script in the repo, like bin2h. if you've achieved this, @rynho, might also consider creating a PR. ;)

neo305, 2024-02-29:
How do I update the update.h file? Is that file what version 1.05 is?[image]

stooged, 2024-02-29:
> How do I update the update.h file? Is that file what version 1.05 [...]

Thats a modification to my original code so i dont know what they have done.
It would be best to contact the person releasing that and ask them.

closed 2024-02-29 by stooged

neo305, 2024-02-29:
Thanks. What would I be asking for? The Update.h file? Or the Pages.h file?

stooged, 2024-02-29:
> Thanks. What would I be asking for? The Update.h file? Or the Pages.h file?

I dont know what updates you are talking about.

neo305, 2024-02-29:
The admin page I guess
[image]

stooged, 2024-02-29:
Not sure, i dont release compiled updates with firmware version increments i only post the code.
What people do with it after that is their choice and you would need to talk to them about issues with their forks of the code.

neo305, 2024-02-29:
Nah no issue. I want his modded code and he won't give it up lol can you ask him for it? Maybe he'll give it to you lol

stooged, 2024-02-29:
> Nah no issue. I want his modded code and he won't give it up lol can you ask him for it? Maybe he'll give it to you lol

Well if he is using other peoples code and not posting the modified code it says alot, i wouldnt use it myself because there is only one reason i can think of to why someone would take open source, modify it and keep it private.