stooged/ESP32-Server-900u #98: No Wifi AP

Killaker, 2023-03-07:
Hi, i have uploaded to my ESP32 S2 mini, without any errors using Arduino IDE, pressed RST button when finished the upload but there's no any access point showing, whats the problem?

Thanks

neo305, 2023-03-07:
You don’t hit reset. You just unplug it.

Killaker, 2023-03-08:
[image]

Killaker, 2023-03-08:
[image]

Killaker, 2023-03-08:
[image]

Killaker, 2023-03-08:
[image]

Francesco146, 2024-02-10:
have you uploaded the files? did you put it in flash mode?