stooged/ESP32-Server-900u #100: ESPAsyncWebServer.h: No such file or directory installing library from IDE

choldi, 2023-04-17:
I was getting the same error as  #63 and I realized that I'd installed the library from IDE instead of using the zip from github

If you add the ESP32AsyncWebServer from Arduino ide 2.04, the library is renamed as ESP32AsyncWebSrv.
You have to change
`#include "ESPAsyncWebServer.h"`
to 
`#include "ESPAsyncWebSrv.h"`

and it compiles right and it works perfectly

stinger101mg, 2023-08-22:
i will attest to this i have an esp32 s3, after i did this and had all the libraries installed and working the sketch compiled perfectly and seems to operate as it should. i have yet to test it on my ps4 but i expect there to be no issues.

Francesco146, 2024-02-10:
consider following the readme. i installed the core library through the ide and then the 2 zips. i had no issues.

consider closing this :)