stooged/ESP32-Server-900u #102: UsbWaitTime not working

eemcm2, 2023-10-15:
when setting another value in UsbWaitTime both in the sketch and in the administration page, it does not work, the value of 10000ms is kept

Francesco146, 2024-02-10:
i think you're not uploading it correctly. note also that the value of the `usb wait` will be limited in this range:
[5000, 10000]

eemcm2, 2024-02-12:
> Creo que no lo estás subiendo correctamente. tenga en cuenta también que el valor de `usb wait`will estará limitado en este rango: [5000, 25000]

Francesco146, 2024-02-12:
could you provide more info? reopening the issue without further details is not really helpful

eemcm2, 2024-02-12:
> could you provide more info? reopening the issue without further details is not really helpful

I have not been able to get usb wait to work, I have changed the configuration in the sketch, I also changed the following:

```
void handlePayloads(AsyncWebServerRequest *request) {
  File dir = FILESYS.open("/");
  String output = "<!DOCTYPE html><html><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1\"><title>ESP Server</title><link rel=\"stylesheet\" href=\"style.css\"><style>body { background-color: #1451AE; color: #ffffff; font-size: 14px; font-weight: bold; margin: 0 0 0 0.0; overflow-y:hidden; text-shadow: 3px 2px DodgerBlue;}</style><script>function setpayload(payload,title,waittime){ sessionStorage.setItem('payload', payload); sessionStorage.setItem('title', title); sessionStorage.setItem('waittime', waittime);  window.open('loader.html', '_self');}</script></head><body><center><h1>9.00 Payloads</h1>";
  int cntr = 0;
  int payloadCount = 0;
  if (USB_WAIT < 2000) { USB_WAIT = 2000; }  // correct unrealistic timing values
  if (USB_WAIT > 5000) { USB_WAIT = 5000; }
```

It still doesn't work, it stays slow when loading exfathax

Francesco146, 2024-02-12:
what value did you give it? what value do you record between the end of the webkit exploit and the beginning of exfatax?

stooged, 2024-02-13:
i have tested the function and it is working as intended, [the values you are playing with](https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L341) are a sanity check and not where you would set the desired wait time.

if you want to set the time and you have the [config enabled](https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L53) then you use the admin.html page to set the wait time.

if you have the config disabled for hardcoded values only then you set the [wait time here](https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L73).

if you wanted to have the wait time less than 5 seconds or more than 25 seconds you would then edit that [sanity check](https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/ESP32_Server_900u.ino#L341).

the reason i chose 10 seconds before executing the exfathax exploit is if you attempt to rush this process it will make the system unstable and panic.
the system needs time to process the corrupt filesystem header information for the exploit to work correctly.

there is no benefit in saving 5 seconds on the load of the exploit when you are doing it one time per console boot and the only reason i added the wait time function was if someone was having stability issues and needed to wait longer than 10 seconds.

with psfree the loading process is quicker now.

if you want to remove the wait time totally you need to unpack the [loader_gz](https://github.com/stooged/ESP32-Server-900u/blob/main/ESP32_Server_900u/Loader.h) and edit the following line
```sleep(usbWaitTime).then(() => {```

you can change `usbWaitTime` to a value like '1000' and the wait time code in the esp will be totally ignored.

closed 2024-02-13 by stooged

Francesco146, 2024-02-13:
> the reason i chose 10 seconds before executing the exfathax exploit is if you attempt to rush this process it will make the system unstable and panic.

I've made dozen on tests and found out that `3000` is the lowest value for which the system does not crash. not sure if it depends to the model of the PS4 and the current memory stack.